prampec/IotWebConf #124: On ESP32 the password is reset after reboot

lamello, 2020-05-22:
Hello,

On my ES32 there is a strange behaviour of the library. Every time after a reboot the _forceDefaultPassword is active (without pressing the button).

I've found a solution.

I have changed this: (IotWebConf.cpp line:119 en up)
` 
if (this->_configPin >= 0)
  {
    pinMode(this->_configPin, INPUT_PULLUP);
    this->_forceDefaultPassword = (digitalRead(this->_configPin == LOW));
  }
`

for this:
`
 if (this->_configPin >= 0)
  {
    pinMode(this->_configPin, INPUT_PULLUP);
    digitalRead(this->_configPin);  // dummy read for timing purposes
    this->_forceDefaultPassword = !digitalRead(this->_configPin);
  }
`

Thanks for sharing this library

prampec, 2020-05-22:
So it basically means, that ESP32 needs some time after pinMode is set and the first digitalRead is performed.
It is good to know. Thank you for pointing that out!

cdaller, 2021-01-23:
I can confirm the problem on an ESP32 and also confirm the fix:
```
bool IotWebConf::init()
{
  // -- Setup pins.
  if (this->_configPin >= 0)
  {
    pinMode(this->_configPin, INPUT_PULLUP);
#if defined(ESP32)
    digitalRead(this->_configPin); // #124: dummy read for timing purposes < ==== this fixes ESP32 problem
#endif
    this->_forceDefaultPassword = (digitalRead(this->_configPin) == LOW);
  }
```

I thought about a PR for this fix but then I saw that there is a branch named "possibleESP32fix".
@prampec how is the status for this?

Thanks for this library: I tried to do a similar library but could not get the loading/saving of the params, so I switched my first project to yours! But as it is running on an ESP32, I needed this patch.

prampec, 2021-04-14:
Sorry for your abandoned issue, I will address this one soon.

theelims, 2021-07-22:
Hi, I just stumbled across this bug as well. It is still persistent in the latest release. However, the bugfix is effective.

theelims, 2021-07-26:
I found an other workaround, that should be transparent to the library. Instead of placing an additional digitalread inside ofIoTWebConf.cpp I placed `pinMode(BUTTON_PIN, INPUT_PULLUP);` just in front of calling `iotWebConf.init();`.

lamello, 2021-07-29:
With all respect, that's not an example of good programming.
With your proposel the init function is depending of code outside of that function if it works reliable.

The solution of cdaller is the correct method.

theelims, 2021-08-01:
> With your proposel the init function is depending of code outside of that function if it works reliable.
> 
> The solution of cdaller is the correct method.

Fixing the code inside the library is certainly the better way.