prampec/IotWebConf #125: Not working with DMD2 library

anees042, 2020-05-22:
Hi, 

I was working on dot matrix display, DMD2 works fine separately, IotWebConf works fine separately but when i try to combine both, DMD starts flickering and it looks like its restarting every second, also for the status LED.

`// not working code
// Insert file Library 
#include <SPI.h> 
#include <DMD2.h> 
#include <fonts/Arial_Black_16.h> 
#include <IotWebConf.h>

SPIDMD dmd(1,1); // Number of P10 panels used X, Y 
DMD_TextBox box(dmd, 2, 1, 32, 16); // Set Box (dmd, x, y, Height, Width) 

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "STDisplay01";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "object00";

DNSServer dnsServer;
WebServer server(80);

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword);

void setup() { 
  dmd.setBrightness(10); // Set brightness 0 - 255 
  dmd.selectFont(Arial_Black_16); // Font used dmd.begin(); // Start DMD 
  dmd.begin();
  
  box.print("001"); // Display TEXT 001
  dmd.drawBox(31, 15, 0, 0); //{x,y,} // drawing box bording led pannel
  box.clear();
  box.print("000"); // Display TEXT 000
  dmd.drawBox(31, 15, 0, 0); //{x,y,} // drawing box bording led pannel

  // -- Initializing the configuration.
  iotWebConf.init();

  // -- Set up required URL handlers on the web server.
  server.on("/", handleRoot);
  server.on("/config", []{ iotWebConf.handleConfig(); });
  server.on("/msg", handleMsg);
  server.onNotFound([](){ iotWebConf.handleNotFound(); });

}

void loop() {
  // -- doLoop should be called as frequently as possible.
  iotWebConf.doLoop();
}

/**
 * Handle web requests to "/" path.
 */
void handleRoot()
{
  // -- Let IotWebConf test and handle captive portal requests.
  if (iotWebConf.handleCaptivePortal())
  {
    // -- Captive portal request were already served.
    return;
  }
  String s = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/>";
  s += "<title>IotWebConf</title></head><body>Welcome!";
  s += "Go to <a href='config'>configure page</a> to change settings.";
  s += "</body></html>\n";

  server.send(200, "text/html", s);
}

void handleMsg()
{
  String message = "";
  if (server.arg("message")== ""){     //Parameter not found
    message = "Message Argument not found";
  }else{     //Parameter found
    message = "Message Argument = ";
    message += server.arg("message");     //Gets the value of the query parameter
    box.clear();
    box.print(server.arg("message")); // Display TEXT 000
    dmd.drawBox(31, 15, 0, 0); //{x,y,} // drawing box bording led pannel
  }
  server.send(200, "text/plain", message);          //Returns the HTTP response
}`

prampec, 2020-05-22:
First try your code against a standard WebServer example.
IotWebConf is basically a smart wrapper around WebServer and mDNS.
You might also want to disable "IOTWEBCONF_STATUS_ENABLED" and remove yield() from line 556 as it is there just to be on the safe side.

anees042, 2020-05-27:
same result with standard WebServer example. i think problem is with DMD2 library.

closed 2020-12-20 by prampec