prampec/IotWebConf #126: EEPROM: Initialization is wrong

jsolla, 2020-05-24:
In current code in the method IotWebConf::configInit() there is this code:

```
EEPROM.begin(
      IOTWEBCONF_CONFIG_START + IOTWEBCONF_CONFIG_VERSION_LENGTH + size);
```
EEPROM.begin() method is supposed to receive the amount of bytes to be accessed, so in case someone changes the IOT_WEBCONF_START offset value (value different than 0) it will allocate some weird value.

It should be:
```
EEPROM.begin(
      IOTWEBCONF_CONFIG_VERSION_LENGTH + size);
```

Miau09, 2020-05-26:
I have had (or still have) some issues (#121 ) because I am also accessing the eeprom for my own stuff.
But as far as I understood, eeprom.begin creates an internal buffer, with the size I define, and copies the files from the emulated eeprom in flash, into this buffer.
The way I understood it, is that it always starts at 0. So if I have an offset of 250, and set a size of 4, how is the begin function supposed to know where I want to start reading? So I would need to call begin(254).

I have an offset of 150, and I fixed most of my issues by simply calling a begin(512) in my eeprom access functions (which includes the maximum length that the IoT lib is using in my sketch).

jsolla, 2020-05-26:
Please check ESP8266 reference for the EEPROM library:
https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#eeprom

Taking into account the fact that the EEPROM library from the ESP8266 has been ported to the ESP32, we can assume behavior is the same.

I paste here relevant text about EEPROM.begin():
`You need to call EEPROM.begin(size) before you start reading or writing, size being the number of bytes you want to use. Size can be anywhere between 4 and 4096 bytes.
`
About your question:
> how is the begin function supposed to know where I want to start reading?

The begin function was never meant to know when you want to start reading, this is why you have a EEPROM.read(ADDRESS) function so you can pass the address you want to read. And the same goes for writes.

Miau09, 2020-05-28:
I am sorry but I cannot aggree, I have just tested it, and this is not working:
EEPROM.begin(4);
value = EEPROM.read(6);

I read zero, but if I define EEPROM.begin(8), I read 69, which is correct.

There is no good documentation about this EEPROM, the only one I found was this:
https://ullisroboterseite.de/esp8266-EEPROM.html

It is in german, but the picture is enough imho.

jsolla, 2020-05-28:
When you are using EEPROM.begin(4) basically you are saying "I would like to operate in the eeprom from address 0 to address 3 (4 bytes)" So if you read address 6 you are out of boundaries, so of course it doesn't work.

So when you use EEPROM.begin(8) you can operate in addresses 0 to 7 therefore it works when you read address 6.

jsolla, 2020-05-28:
Im closing this issue cause I realized incrementing the offset will actually produce the desired effect. So not an issue :)

closed 2020-05-28 by jsolla