prampec/IotWebConf #128: EEPROM: IOTWEBCONF_CONFIG_START cannot be "easily" changed

jsolla, 2020-05-24:
The provided method to add offset value to the storage area of the parameters in EEPROM is done via a #define this way:

```
#ifndef IOTWEBCONF_CONFIG_START
# define IOTWEBCONF_CONFIG_START 0
#endif
```
Turns out, the way Arduino framework is created makes it very almost impossible to override this definition in a clean way. For example adding a **#define IOTWEBCONF_CONFIG_START <value>** to your .ino file wont work because the library is compiled separatedly and linked later on.

Other than modifiying the library, there is no clean way to fix this situation (AFAIK).

If the user application (or another library) needs to make use of the EEPROM it is impossible to allocate a reserved area for it as IotWebConfig will override this area.

Proposal is to add a setConfigStart(offset) method to let the user application setup the offset address before init() is called. 

I have a working POC if you are interested, just let me know and I will submit a PR.

Miau09, 2020-05-26:
I would be interested to know how you fixed this.
I also need this offset, and have simply modified the library file for now.

But I know it is possible to do it like you described, I have seen it for example in the task scheduler library

jsolla, 2020-05-26:
@Miau09 I don't know exactly what you mean by the example on the task scheduler, could you plase clarify?

Miau09, 2020-05-27:
https://github.com/arkhipenko/TaskScheduler/blob/master/src/TaskSchedulerDeclarations.h

At the beginning you can find the defines, and I can use them in my main program, like this:
#define _TASK_TIMECRITICAL
#define _TASK_SLEEP_ON_IDLE_RUN
#include <TaskScheduler.h>

Not sure how this is done, and if it is also working with some values.
Unfortunately I am not that much of an expert, don't know if this can help.

jsolla, 2020-05-27:
Now I see what you mean.
In the case of that TaskScheduler library it works because the whole library is declared in a .h file, therefore is not compiled separatedly but along with your .ino file so this is why your #defines work.

Unfortunately this is not the case of iotwebconfig, so the approach of declaring #define statements before including the library is not valid. At least not in my tests, and not according to many people in the forums asking the same question.

maf1, 2020-06-06:
This would definitely be a feature I'd need.
An alternative would be to create a new parameter type "blob" or "binary" that is not displayed on the config page and dumped in hex in log output.
(I've not yet tried it, but this should already nearly work (with the exception of the log output) for parameter type "text" setting visible to false).

maf1, 2020-06-06:
As about the proposed setConfigStart(offset) method: 
it would be useful to add a second argument that controls the usage of the EEPROM singleton. 
In principle 3 modes are possible: 
1. "Lazy": Init EEPROM once in init() and keep it alive (current implementation)
   (Together with the offset parameter Apps can directly read, write and commit without calling EEPROM.begin() and end(), but the memory is blocked)
2. "Optimized": Only Init EEPROM when needed (config) and end it afterwards.
   (Apps might do the same to acmes their segment (offset) - memory usage is optimized)
3. "External": Do not call EEPROM.begin() and end(), but let the App do that 
   (App needs to know the size required by iotWebConf parameters, new public method needed)
That plus the new parameter type mentioned above should cover all possible integration scenarios.

jsolla, 2020-06-06:
On my PoC I did 2 things:
1. Add the setConfigStart() method.
2. Add a "OnEEPROMOperation(type, event)" callback registration method: When this callback is registered, IoTWebConf library reports to the other parts of the application about the begin/end calls either for read or write to the eeprom. So from other parts of the application you can use some kind of "mutex" or whatever you need to prevent simultaneous access to the eeprom.

Anyway... for me one thing is clear, the whole eeprom approach on this library needs to be improved, otherwise it is almost impossible to use the eeprom from other parts of the application in a "safe" way.

It seems like the original author is not very active, so not much room for discussion. Might be we should fork it? :)

maf1, 2020-06-07:
I agree that the EEPROM handling is far from being satisfactory. I hope the author picks up some of the features proposed here to make this lib even more valuable, without branching into another version.
For my purposes the trick described above of using a hidden parameter is working.

prampec, 2020-08-17:
There are plenty configuration options on coded into defines are IotWebConf.h .
For the time being I do not see the point extracting all these setting as variables and use setters to control them runtime.
It is a pity not having Arduino control these options compile time, I just hope, that some days they will fix this.
But any way, if you feel yourself professional enough to tune these options, I encourage you to switch to e.g. PlatformIO, where you can already have this.

Btw. EEPROM.end issue is fixed, and will be available from v2.3.2 .

closed 2020-08-17 by prampec