prampec/IotWebConf #129: EEPROM: EEPROM.end() is never called

jsolla, 2020-05-24:
EEPROM.end() needs to be called in order to deallocate the RAM memory allocated during EEPROM.begin().

SymbolCsaba, 2020-06-08:
EEPROM.end() = EEPROM.commit() + free mem. Really good remark, but only 512-4k byte used. And not guarantied to read from EEPROM again...
Do EEPROM.end() in iotWebConf.setConfigSavedCallback(&configSaved) callback!

jsolla, 2020-06-08:
If you check issue #128 I already proposed at least 2 callbacks, one when the library calls begin() and another one when it calls end(). You also need to know when the library is going to access the eeprom to prevent other parts of your application to try to access it in the meantime, specially with a multithreaded OS.

So it should be something like:
iotWebConf.setCofigBeginSaveCallback(configSaveBegin);
iotWebConf.setConfigEndSaveCallback(configSaveEnd);

jsolla, 2020-08-11:
@prampec can we consider this project like...abandoned?

prampec, 2020-08-11:
I wouldn't say that. The thing is that this is a one-man army, so basically I fix things, that I'm also facing with. And currently I do not work too much with the library.
About this one...
I do not see what would a "multithreaded OS" is about to mean in ESP8266 and even ESP32 context.

jsolla, 2020-08-11:
Ahh glad to see you are around!

**I do not see what would a "multithreaded OS" is about to mean in ESP8266 and even ESP32 context.**
Well, the whole arduino framework is just running into FreeRTOS thread, at least on the ESP32.

Btw, I can provide the mentioned code if you are happy to accept PRs.

prampec, 2020-08-12:
I was investigating FreeRTOS, but I'm still not convinced, that EEPROM handling needs callbacks. IotWebConf does EEPROM manipulations in one go (one method call). I do not see any sign of the FreeRTOS to have the ability to interrupt a running method call and start another one, then continue the interrupted call. That would expect the CPU to have context switching option. I'm not sure about this, but context switching in small MCUs like ESPs would too expensive, and I do not believe, they support it at all.

jsolla, 2020-08-13:
FreeRTOS is doing context switching based on a hardware timer, of course, and there are threads working without any doubt. Like it or not is how the Arduino environment is working on the ESP32.

So yes, your code calls eeprom.begin() at some point and writes later on. This approach basically prevents any other library to use the eeprom in a proper way without producing side effects, even if no RTOS was present on the code.

prampec, 2020-08-13:
Okay. I totally understand the lack of EEPROM.end(). This is a bug and it needs to be fixed.

jsolla, 2020-08-15:
Please take into account is not only not calling end. Consider performing the operation in one go, something like this pseudocode:

```
// Call eeprom begin callback, so user code can avoid concurrent access
if (cbEEPROMBegin) cbEEPROMBegin()

// Call begin
EEPROM.begin(params here...)

// Perform all required write operations
EEPROM.write(params here....)

// Finally write.
EEPROM.end()

// Call EEPROM end callback, so user code can release locks or whatever it was using
if (cbEEPROMEnd cbEEPROMEnd()
```
As you can see something simple like adding callbacks to tell the user whenever an operation is going to be performed would be enough to avoid collisions and removes the conflict.

prampec, 2020-08-17:
Fixed. Will be available from v2.3.2 .

closed 2020-08-17 by prampec