prampec/IotWebConf #130: Feature request: Reset password in software, without using ConfigPin

societyofrobots, 2020-05-25:
My ESP32 will not have direct physical access, but will occasionally need a password reset.

How can I do this without using the ConfigPin?

societyofrobots, 2020-08-24:
After a few days of hair pulling, I hacked up a mod to add this feature.

In your main code, call these two lines when you want to reset the password:
```
      iotWebConf.pass_reset();
      needReset = true;
```
In IotWebConf.h,
        before:       
`         boolean init();`
        
        add:
`         void pass_reset();`

In IotWebConf.cpp:
        before:
`          void IotWebConf::init()`
          
        add:
```
          void IotWebConf::pass_reset()
          {
            int size = this->configInit();
          
            EEPROM.begin(IOTWEBCONF_CONFIG_START + IOTWEBCONF_CONFIG_VERSION_LENGTH + size);
            for (uint8_t t = 0; t < IOTWEBCONF_PASSWORD_LEN ; t++)
              EEPROM.write(IOTWEBCONF_CONFIG_START + t, this->_apPassword[t]);
            EEPROM.end();
          }
```

prampec, 2020-08-25:
I still don't see the use-case of this feature.
1. You have the forceApMode(), that will put the thing into AP mode runtime.
2. You can access built in config parameter values directly e.g. getWifiSsidParameter()
3. You can save modified configuration with configSave()

What you might still miss is the option to force default password runtime (what exactly the configPin does on startup). But then you need a setter for the _forceDefaultPassword parameter.

I do not recommend to mess around by modifying values EEPROM values directly.

societyofrobots, 2020-08-25:
I will not have physical access to my ESP32, but will occasionally need to remotely reset password to default. This rules out configPin. It's also only run in AP mode, which I think rules out forceApMode().

Perhaps I'm not understanding it right, but does forceApMode(), getWifiSsidParameter(), and/or configSave() reset the AP password to default?

prampec, 2020-08-25:
The configPin is used to force default password on startup and enter/stay in AP mode. The configPin option does not change any stored configuration automatically. In the title you have suggested a feature to do this from the code. This does not have too much sense, this is why I didn't have answered this topic so far.

Now it turns out, that you want something else. Maybe you want to change the stored configuration. You can do that by e.g. getWifiSsidParameter() strncpy the new value to the valueBuffer of the IotWebConfParameter.
If so, you also need to use the configSave() method to store this change in EEPROM.

societyofrobots, 2020-08-26:
All I want is a software command that remotely resets the AP password to the initial stored default password. It's in a remote location and I don't have the ability to touch it.

> You can do that by e.g. getWifiSsidParameter() strncpy the new value to the valueBuffer of the IotWebConfParameter. If so, you also need to use the configSave() method to store this change in EEPROM.

I'm not a very strong programmer and that was Greek to me haha. It took me a week of pulling my hair out just to figure out the dirty EEPROM hack.

I'm assuming it involves something like this but I don't know where to go from here:

```
iotWebConf.IotWebConfParameter("AP password", "iwcApPassword", this->_apPassword, IOTWEBCONF_PASSWORD_LEN, "password");
configSave();
```

prampec, 2020-08-27:
Okay! Let's do it step by step.
Any time you need to change the AP password from the software, you need to do the following:

```
IotWebConfParameter* wifiPasswordParameter = iotWebConf.getApPasswordParameter();
strncpy(wifiPasswordParameter->valueBuffer, newPassword, IOTWEBCONF_WORD_LEN);
iotWebConf.configSave();
```
In your case newPassword is wifiInitialApPassword.

BUT: I still don't see, why anyone wants to do this. There is no example for this workaround just because I thing this is not the way things should work. You can do it, but you really must know what you are doing. I truly believe, that you have a design issue in your product, this is why I ask the use-case instead of the problem you are willing to solve. Further more....

WARNING: It is a security issue, if you have an option to reset AP password from software to a default one. In this case, if someone figures out your default password, might be able to reset any of your devices, and take control of them. (No wonder I strictly recommend to do this with a hardware button.)

societyofrobots, 2020-08-28:
Thanks! I'll give it a try soon and report back.

> I still don't see, why anyone wants to do this. There is no example for this workaround just because I thing this is not the way things should work.

My ESP32 is in a very secure location where no physical access is possible, and it controls a security door lock. The configPin is a huge security vulnerability, so no user can be allowed access to it.

To resolve the problem of a forgotten password, I am using a secure fingerprint sensor in place of the configPin. When an approved finger is swiped, a software command resets the password. It's actually working right now, but I'd rather do it properly in code and not via my cheap hack.

prampec, 2020-08-28:
Okay! Now I see, why you need this. So you have got the solution above.
The only thing you need to take care, is that the software code you are using for resetting the password must not be controlled from the network. So this code must be access by hardwired internal logic.

societyofrobots, 2020-10-02:
Reporting back that it worked without issue.

Question, how do I read the AP password that's stored in ram? I want to use it as the BLE pin key.

For example:
uint32_t BLE_pinkey = atoi( AP_password_in_ram ) ; //atoi() for changing char array to int

prampec, 2020-10-02:
Same thing as mentioned before.

closed 2020-10-02 by prampec

societyofrobots, 2020-10-05:
It wasn't quite the same thing as mentioned before, but I eventually figured it out:

```
IotWebConfParameter* wifiPasswordParameter = iotWebConf.getApPasswordParameter();
uint32_t BLE_pinkey = atoi(wifiPasswordParameter->valueBuffer);
```