prampec/IotWebConf #133: Feature Request: Async TCP

jlgoasis, 2020-06-24:
It would be nice to add an example with Async TCP connection

Any idea how do that?

Thank you

tnagels, 2020-09-11:
I have the same question. I have tried adding a telnet server using some other examples but cannot get it to work, the esp goes in a reboot-loop...

h2zero, 2020-09-14:
I have created PR #142 to address this. If you test it please post feedback there.

prampec, 2021-04-14:
With v3.0.0 you are free to add alternative web-server (in this case async-web-server) instead of the regular one.
Unfortunately currently I do not have the resource to do this, but it is open to the public to do. I would be very happy to accept a pull-request with addition from someone.

closed 2021-04-14 by prampec