prampec/IotWebConf #136: Not loading the html

antonioreyna, 2020-07-17:
Hello sometimes when i load the config webpage, it does not load the getFormStart, i need to reboot the device and then it shows the html correctly.

jvondrus, 2020-11-22:
Hello, I have a similar problem.
Web page is generated and loaded, but did not contains all fields. Sometimes some "fieldset" missing and sometimes missing some rows in that. Rendering size is random.
My config page has 8 fieldset. First one is default (system settings), next 7 is my fields, each one contrails like 5 values elements.

It's looks that problem is at generating web page contains string (page and pitem), because source code of page is finished property... "Firmware version 'xxx'</div></div></body></html>".

jvondrus, 2020-11-22:
And sometimes it crash and restart. Here is a dump for that crash:

`Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
PC: 0x4000df64
EXCVADDR: 0x00000000

Decoding stack results
0x40100824: pvPortZalloc(size_t, char const*, int) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\heap.cpp line 301
0x402275cc: loop_task(ETSEvent*) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 205
0x401003d1: __wrap_spi_flash_read(uint32_t, uint32_t*, size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_phy.cpp line 309
0x402247a5: EspClass::flashRead(unsigned int, unsigned int*, unsigned int) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Esp.cpp line 733
0x40228bf0: flash_hal_read(unsigned int, unsigned int, unsigned char*) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\flash_hal.cpp line 67
0x40231856: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40100380: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40231856: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40100380: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40234995: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40100ef4: malloc(size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x40230e79: _Balloc at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 124
0x40100ef4: malloc(size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x4023504c: ethernet_output at netif/ethernet.c line 312
0x40101144: calloc(size_t, size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 902
0x40230e79: _Balloc at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 124
0x40100ef4: malloc(size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x402315b2: __d2b at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 786
0x40101152: calloc(size_t, size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 908
0x40101144: calloc(size_t, size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 902
0x40231856: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x40234995: glue2esp_linkoutput at glue-esp/lwip-esp.c line 301
0x40231c87: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 667
0x40234c22: new_linkoutput at glue-lwip/lwip-git.c line 260
0x4023504c: ethernet_output at netif/ethernet.c line 312
0x40100c3b: umm_free_core(void*) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x4022d615: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4023c81c: etharp_output_to_arp_index at core/ipv4/etharp.c line 770
0x40100c3b: umm_free_core(void*) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 351
0x4023ca70: etharp_output_LWIP2 at core/ipv4/etharp.c line 885
0x4023e370: ip4_output_if_opt_src at core/ipv4/ip4.c line 1764
0x40100ef4: malloc(size_t) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x4023ee60: mem_malloc at core/mem.c line 210
0x4023e3b8: ip4_output_if_opt at core/ipv4/ip4.c line 1577
0x40239691: tcp_create_segment at core/tcp_out.c line 190
0x4023e3de: ip4_output_if at core/ipv4/ip4.c line 1550
0x4023ef6b: ip_chksum_pseudo at core/inet_chksum.c line 395
0x4023a3a6: tcp_output at core/tcp_out.c line 1361
0x4022764c: __esp_yield() at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_features.h line 92
0x4022860a: __delay(unsigned long) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_wiring.cpp line 54
0x4021c8ff: ClientContext::_write_from_source(DataSource*) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/include/ClientContext.h line 477
0x4021c989: WiFiClient::write(unsigned char const*, unsigned int) at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/include/ClientContext.h line 377
[... cut ...]
0x4021ac93: esp8266webserver::ESP8266WebServerTemplate ::handleClient() at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 340
0x4022768e: __yield() at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 112
0x4021adad: IotWebConf::doLoop() at C:\Users\XXXX\Documents\Arduino\libraries\IotWebConf\src\IotWebConf.cpp line 821
0x4020e7dd: loop() at E:\Thermo_v16/Thermo_v16.ino line 87
0x40227770: loop_wrapper() at C:\Users\XXXX\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197`

prampec, 2020-12-01:
I have just found a bug, that is very likely to cause trouble on ESP32 (but doesn't noticed problem in ESP8266). It is already fixed in branch CustomParameters, but it is very likely, that it needs to be fixed in the main line ASAP as well.

jvondrus, 2020-12-13:
Do you can tell us how to fix it? There are a lot of changes at branch CustomParameters...

prampec, 2020-12-13:
I just have released v2.3.3. Please check!

jvondrus, 2020-12-13:
Release 2.3.3 didn't help with that, but branch CustomParameters **did**.

prampec, 2020-12-13:
Weird. Okay. I'm  just about to release a new branch with some more
updates, fixes, etc. based on CustomParameters. So please don't rely on the
CustomParameters branch any more, take it as a temporary solution.

prampec, 2020-12-13:
Meanwhile I have made yet another modification with a potential bugfix. Please try branch "possibleESP32fix"! If this one fixes your issue, that I will immediately release a new 2.3.4 with it.

jvondrus, 2020-12-14:
Didn't fixed :(
Branch **CustomParameters** works perfect, also new mode **IotWebConfSelectParameter** works, it's a really good function :)

prampec, 2020-12-14:
Then I don't know what fixed the problem in CustomParameters.
So, I need to speed up development get a reliable branch ASAP.

jvondrus, 2020-12-14:
It's looks, that problem was solved with new code at "IotWebConf.cpp" for function:
`IotWebConf::handleConfig()`

prampec, 2020-12-15:
Huhh... Lot has changed in that method.

prampec, 2020-12-18:
Please use "dev" branch instead of CustomParameters further on.
The dev branch will be the new release branch very soon. No major changes are planned on this branch.

jvondrus, 2020-12-18:
Heh, a lot of changes :)
New debug of parametr looks good :)

Html works perfect, but it requires much more memory (heap). Around more than 3k. But still works :)
Great work :+1:

prampec, 2020-12-20:
I have made v3.0.0 as the default master branch. So all changes here are official from now on.

jvondrus, 2021-01-05:
Hello, v3.0.0 works perfect, thank you.
I think, you can close this :)

prampec, 2021-01-06:
Thank you for your feedback!

closed 2021-01-06 by prampec