prampec/IotWebConf #139: how to best add encryption for over the internet connection?

societyofrobots, 2020-08-11:
AP mode is encrypted, however the documentation says it loses encryption once it goes through a router and onto the internet.

Can someone direct me to links or conceptually describe what I need to do to best encrypt the internet connection?

I'm an average skilled programmer, so nothing too complex please!

prampec, 2020-08-11:
This is a good question.
The internet connection just cannot be encrypted. All devices on the same network can view all traffic. What you can do is to set up a dedicated network (e.g. set up a wifi router for your IoT devices), or to encrypt your traffic. In our case you might want to set up a secure web server, as shown in this article:
https://www.onetransistor.eu/2019/04/https-server-on-esp8266-nodemcu.html

The problem here, is that your encryption keys are not signed by a trusted authority, but you sign it yourself.
But before solving this problem, you need to ask yourself who is going to connect to your server: Is this a public access server, is it local area server accessed by internal users, or is it a local server accessed by a very limited browsers.
For the last case, importing self-signed certificate to the browser can be enough, as shown in the article above.
 
When we are talking about a public accessible server on the Internet, with self signed certificates 3rd party user cannot be sure, that the connection is really set up to your server, or someone is imitating your server. Thus you need to sign your keys with trusted authorities, as shown here:
https://medium.com/@flynam/securing-your-iot-device-using-ssl-4643110ab901

If you want to set up a server on local network, where any random local user might want to access your server, then you are in trouble. You need to set up a local certificate authority to sign the keys, but you also want to sign the certificate of your authority by a trusted one. And this typically not an easy task costs much.

Also note, that certificates needs to be refreshed regularly, thus your device needs to be maintained regularly.

Search keywords: esp8266 https server letsencrypt

societyofrobots, 2020-11-10:
I still haven't quite solved this problem, but I'm putting this here for others who are interested.

The two best methods I've found are...

OAuth2 for Google, which automatically provisions certificates:
https://developers.google.com/identity/protocols/oauth2

and Amazon AWS Provisioning for Devices, which does the same.
https://docs.aws.amazon.com/iot/latest/developerguide/provision-wo-cert.html#claim-based

Unfortunately I couldn't find any Arduino libraries to do either, and my coding ability isn't at the level needed to do this myself.

prampec, 2021-04-14:
There is a way to set up secure web server for the ESPs. It still needs infrastructure (keys and whatnot) to operate. From version 3.0.0 we have on option to use alternative web server (in this case secure web server) instead of the regular one.
Unfortunately I do not have the resources to implement this option, but it is open to the public to do. I would be happy to accept this pull request from someone.

myqel, 2022-09-14:
Hopefully it is ok to append this thread here.
I managed to get an ESP32 working with self signed certficates to a MQTT server over TLS. I added a parameter type MultiLineTextParameter with HTML tag <textarea> to IoTWebConfParameter to handle CA.crt, ESP.crt and ESP.key (This would realize both side security, with CA.crt only it could be an authentification of the server only alternative) via the captive portal and easily copy&paste the certificates in a web browser. Now I stuck on the EEPROM 4k limitation, as these multiline strings consume this space completely or they even do not fit into it (EEPROM.commit() fails). The byte stream you handle is programmed greatly, but it will be a big effort to change the concept to ESP32 NVS Preferences (I guess it would even simplify the code a lot) and it is not compatible to ESP8266 anymore I guess. Do you have a an idea to get a second 4k flash page added and stick on EEPROM.h?

myqel, 2022-09-25:
I managed the EEPROM issue as mentioned and changed the byte stream to single keys in preferences. So it works for me now. This code works for ESP32 only and for sure it is no answer to the question of this thread, so I finish writing here.
Thanks for the great code! Cheers; myqel.
[image]