prampec/IotWebConf #141: Intializing IotWebConf outsize the setup() Vs inside

worthapenny, 2020-09-02:
The block of code below runs fine (where **_IotWebConf is initialized outside the setup()_** method
```cpp
//...
IotWebConf *iotWebConf;
IotWebConf iwc(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
//...

void setup(){
  //...
  iotWebConf = &iwc;
  iotWebConf->setStatusPin(STATUS_PIN);
  iotWebConf->init();
  //...
}
```

Now, the below block crashes. Here the **_IotWebConf class is initialized in setup()_** block
```cpp
//...
IotWebConf *iotWebConf;
//IotWebConf iwc(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
//...

void setup(){
  //...
  IotWebConf iwc(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
  iotWebConf = &iwc;
  iotWebConf->setStatusPin(STATUS_PIN);
  iotWebConf->init();
  //...
}
```

Some output from Serial (When IotWebConf is initialized in the setup() block):
```
Starting up...
Config size: 165
Loaded config 'iwcThingName'= 'testThing'
Loaded config 'iwcApPassword'= <hidden>
Loaded config 'iwcWifiSsid'= 'MyWiFiSSID'
Loaded config 'iwcWifiPassword'= <hidden>
Loaded config 'iwcApTimeout'= '30'
Ready.
----+----+----+----+----+----+----+----+----
Begin Loop
State changing from: 0 to 2
Setting up AP: testThing
Use password: <hidden>

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```

### How can I initialize the IotWebConf class inside the setup() block and prevent the crash?
<ins>I hope being a rookie in c/cpp is not the issue.</ins>

prampec, 2020-09-02:
The problem here, is that variable iwc has scope setup() {} and will be destroyed after setup exits.
What you can do however to declare iwc as const in the method.

'
IotWebConf *iotWebConf;

void setup(){
  //...
  const IotWebConf iwc(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
  iotWebConf = &iwc;
  iotWebConf->setStatusPin(STATUS_PIN);
  iotWebConf->init();
  //...
}
'

(Overall it is not a good idea to store globally used variables as pointers in Arduino. And I personally do not recommend this kind of instantiation.)

Please came back with your results!

worthapenny, 2020-09-02:
with the const compiler gives `a value of type "const IotWebConf *" cannot be assigned to an entity of type "IotWebConf *"`
```cpp
const IotWebConf iwc(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
iotWebConf = &iwc;
```
(I am using PlatoformIO and I get a red squiggly line with the error as shown above)

The reason I am trying to initialize the class inside the setup is that I am moving the contents from the main.cpp to my own library (let's say) like MyLib.cpp/MyLib.h

Thanks for your suggestion on best practices.

prampec, 2020-09-02:
Okay I understand, what you mean, but you even cannot instantiate WebServer, etc. other than in global scope.

Now the biggest concern with Arduino, is that we want to avoid dynamic instantiation (thus creating object instances with the "new" keyword. This would cause runtime memory allocation. But we want to know memory usage on build time. This is why we try to create all instances in global scope.

What I can recommend is to make your library in a way, that it receives IotWebConf reference as a parameter.

(I believe you can simply cast (const IotWebConf *) to (IotWebConf *) or you might want to define (IotWebConf *iotWebConf;) as const as well.)

closed 2021-03-28 by prampec