prampec/IotWebConf #144: Add legend to the first separator?

Klausoleum, 2020-09-29:
Thank you for this wonderful library.

Is there a possibility to add a legend to the first separator?

[image]

jvondrus, 2020-11-05:
Hello, I did it by myself by edit library source.
Find "IotWebConf.h" and edit the line:

`const char IOTWEBCONF_HTML_FORM_START[] PROGMEM   = "<form action='' method='post'><fieldset><legend><b>System</b></legend><input type='hidden' name='iotSave' value='true'>";`

prampec, 2021-01-14:
A completely new approach was introduced with v3.0.0, where you have even more possibilities to customize fieldsets.

closed 2021-01-14 by prampec