prampec/IotWebConf #147: Update FileSystem with data.bin file

jvondrus, 2020-11-27:
Hello,
I'm using [LittleFS][1] as file system, but when I create bin file from data by [mklittlefs][2] and try to upload it via your html servise, its faild. LittleFS is "damaged" and empty :(

Thank you for any help.

[1]: https://github.com/lorol/LITTLEFS
[2]: https://github.com/earlephilhower/mklittlefs

prampec, 2020-11-28:
I'm not completely sure how iotWebConf effects LittleFS. What I can tell you for sure, is that iotWebConf stores it's data on a virtual EEPROM, that is actually also stored on the flash. So this might cause some conflict. But I do not see any evidence for that.

jvondrus, 2020-11-28:
Changing configuration for _iotWebConf_ at EEPROM don't effect _LittleFS_.
Reading or writing data to _LittleFS_ during program running didn't effect _iotWebConf_  EEPROM data.
Saving new _iotWebConf_ at program didn't effect _LittleFS_ flash data.
Uploading _LittleFS_ data via _LittleFS Data Upload_ in Arduino IDE works good and don't effect _iotWebConf_ at EEPROM.

The problem occurs when uploading BIN data via HTTP web servis.
Problem can be at my creating BID file from DATA directory (I do not think) or when _iotWebConf_ saving BIN do FLASH.

Which flash system was your BIN data uploading tested?

prampec, 2020-11-28:
If you are talking about firmware upload, then I can say, that I use the standard HTTPUpdateServer for ESP8266. I don't really know details about this approach.

jvondrus, 2020-11-28:
Firmware upload for sketch BIN works good, problem is with flash data BIN.
I will check HTTPUpdateServer procedure for analyse that problem.

jvondrus, 2020-12-19:
Solved :+1: 
LittleFS required specific parameter on bin builder.

For check parameters: [Filesystem information structure](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#info)
My results:
Size [B]:     1024000 
Used [B]:       40960 
Block:              8192 
Page:                256 
Max Open Files:   5 

So my settings for the [BIN creator](https://github.com/earlephilhower/mklittlefs) is:
`mklittlefs.exe -c data -b 8192 -p 256 -d 1 data.bin`

**I hope, that it will help other users of LittleFS :)**

--
Tested on ESP8266 with ESP8266HTTPUpdateServer.h.

closed 2020-12-19 by jvondrus