prampec/IotWebConf #149: Manually handle WiFi connect and disconnect

Ordspilleren, 2020-12-01:
First, I should describe my use-case: In my application, I am using ESP-Now to receive messages from several nodes. Whenever a message is received, the ESP will connect to WiFi and publish the message to MQTT.
Due to the nature of ESP-Now, I cannot use it and WiFi simultaneously. I therefore need to manage connecting and disconnecting from WiFi manually.

I would like to use IotWebConf to manage settings by having it create an AP when a button is pressed, but not automatically handle anything related to WiFi connections. Is this possible to do at the moment? I have not found any obvious way to do this.
This might also very well be out of scope, but I believe it could be useful for anyone using ESP-Now.

coelner, 2021-01-17:
It would be far more easy to use two esp8266, connect them over UART and have one as espnow receiver and the other one as mqtt transmitter.
But back  to your question, currently I see not option because there is missing something like setState. But you can use the RTC from the esp8266 and switch between espnow and STA mode with a deep sleep reboot to exchange the message over the rtc memory (https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/RTCUserMemory/RTCUserMemory.ino). It is enough space because espnow can just transmit 250bytes payload.

prampec, 2021-03-28:
Sorry, I was busy releasing version 3.0.0, but this idea is worth to consider.
First of all, I really do not remember why changeState() method is private. Probably because not always all state is choseable.
The second thing is that you want a forced-offline mode, that is also doable. We will see...

EricDuminil, 2021-05-17:
If I understood correctly (and that's a really big "if"), the config is read in `IotWebConf::init()`, and connections are only started during `IotWebConf::doLoop()`.

To stay offline, the device could simply call `IotWebConf::init()` (or just `IotWebConfloadConfig()`) in `setup()`, but would not call `IotWebConf::doLoop()` from loop().

Inside loop, you could just call `IotWebConf::doLoop()` when required, e.g. when a button is pressed.

Maybe setting `setWifiConnectionTimeoutMs(unsigned long millis);` to 0 or defining `setWifiConnectionHandler` with an empty lambda could ensure that WiFi is not started again.

You could also use `setWifiConnectionFailedHandler( std::function<WifiAuthInfo*()> func )` to completely disable WiFi, and start ESP-Now again.

prampec, 2021-05-17:
@EricDuminil Yes, these tricks very likely do the job. What I wanted here is an "official" recommended way.

prampec, 2021-06-12:
I have provided to you an OffLine mode. Please check example IotWebConf16OffLineMode.
Solution will be available from v3.1.0

closed 2021-06-12 by prampec