prampec/IotWebConf #151: Wifi SSID issue

firestrife23, 2020-12-06:
I'm having issue with connecting to wifi SSID containing special character:  '

For example if I entered this wifi name: `Jason's wifi`

After saving, it changed to `Jason` because it stripped out `'s wifi`

firestrife23, 2020-12-06:
I also tried the escaping backslash and quotation marks, but it doesn't work.

prampec, 2020-12-06:
Try debugging what is happening. I do not see the reason why special characters should not work.
You post your value from the browser. The WebServer decodes your post, and makes available data for us. In iotWebConf we just use the value provided by the WebServer, I do not know any further processing on this.
(Only ASCII characters are valid, but it is okay here.)

prampec, 2021-04-14:
Sorry, I do not have the time to investigate this issue, but if you have some further details, please fill free to reopen it again!

closed 2021-04-14 by prampec

maxsatula, 2025-05-24:
@prampec let's reopen please.

I had the same issue. An apostrophe in a SSID name gets saved to EEPROM correctly, but does not render well in a generated HTML after config page is reloaded. It should be somehow escaped when substituting values in `IOTWEBCONF_HTML_FORM_PARAM`, to avoid being confused with an apostrophe which terminates `value` string

I went a simpler route myself, that was kinda dirty hack, I replaced surrounding apostrophes in HTML template with quotes, hoping I will never have double quotes in a SSID name (but apostrophes are fine now) (IotWebConfParameter.h:31):
```
   "name='{i}' {l} placeholder='{p}' value=\"{v}\" {c}/>"
```

A better solution would be to escape a value in `pitem.replace("{v}", ...` properly though, to handle both apostrophes and quotes.

@firestrife23 you can use my temporary workaround if you want