prampec/IotWebConf #155: two incorrect characters in System Configuration page v3.0.0

arvdsar, 2020-12-21:
Hi,
On version 3.0.0 when opening the configuration page there are two characters '> that do not belong there.

This is what the page is showing on top:
'>System configuration

HTML Source:
```
            <input type='hidden' name='iotSave' value='true'>

            '>
            <legend>System configuration</legend>
            <div class=''>
```

prampec, 2020-12-22:
Thank you for pointing that out. I will address this soon.

prampec, 2020-12-22:
Fixed.

closed 2020-12-22 by prampec

arvdsar, 2020-12-22:
Thanks! Can you point me out where you fixed it? I've been searching to see how to do it myself, but couldn't find it.

prampec, 2020-12-22:
The corporate was in ParameterGroup::renderHtml
```
content = "'>";
```

Where it should have been +=

For me also took some minutes to find it. ;)