prampec/IotWebConf #156: Parameter improvement ideas

rovo89, 2020-12-21:
First of all: Great library, I really like that it's implemented and documented in such a clean way. Also the new parameter customization is very nice.

It still feels a bit clunky in some places though and I have some ideas in mind that could improve them:
- Avoid long constructors by using something like a builder pattern. This would also make it more extensible and easier to read.
- Add possibility to add item to group (and group to main object) right within the constructor, so no additional code in `setup()` would be required.
- For things like numbers, actually use numeric data type instead of `char*` for the value buffer (might work using templates). This would be a huge improvement, just one example is the AP timeout that you have to convert to int before using it.
- Parameter class for IP addresses, could use `IPAddress` internally based on the previous suggestion.
- Generate HTML IDs automatically, in most cases the developer and user doesn't care.
- Make the parameter object the owner/provider of the value buffer instead of passing a reference to an "externally" (from the lib's point of view) variable. Would avoid defining two things for each variable and passing things like size around. Value could be retrieved via a getter. In case someone wants direct access to the variable, the getter could return the reference (`int&`) which could be stored and used just as if it had been declared in the main program in the first place.

For some of these things I'm already thinking about implementation ideas, others are just "wouldn't it be nice if..." ideas. Before I go into detail or try out a few things, I just want to ask: Are you generally open for such changes? They would certainly break backward-compatiblity again. And would GitHub be fine to suggest some ideas?

prampec, 2020-12-22:
Thank you for your feedback. Please note, that I'm just writing a Hacker's cookbook for IotWebConf, that might clarify some things. About the changes: I usually accept changes that are within the concept of IotWebConf, will not effect existing applications footprint and does not bring in dependencies. Version 3.0.0 is not release yet, so compatibility braking changes might still be considered.

>     Avoid long constructors by using something like a builder pattern. This would also make it more extensible and easier to read.
That is a reasonable opinion.

>     Add possibility to add item to group (and group to main object) right within the constructor, so no additional code in setup() would be required.
Normally I try to avoid contructor-time logic, as I have burnt myself with that already several times. But here also sounds reasonable.

>     For things like numbers, actually use numeric data type instead of char* for the value buffer (might work using templates). This would be a huge improvement, just one example is the AP timeout that you have to convert to int before using it.
Yes, I was designed v3.0.0 to be capable for this. But for backward compatibility I didn't have done it for now. Maybe it should be a new property type e.g. NumericBinProperty. On the other hand I also have another idea, that instead of EEPROM, we could store the configuration in the flash in a JSON file.

>     Parameter class for IP addresses, could use IPAddress internally based on the previous suggestion.
This is definitely something I would accept, if somebody would offer is as a pull request. :)

>     Generate HTML IDs automatically, in most cases the developer and user doesn't care.
I don't know. It is not easy to generate ids. Further more check my idea about JSON export above. There we definitely need ids.

> Make the parameter object the owner/provider of the value buffer instead of passing a reference to an "externally" (from the lib's point of view) variable. Would avoid defining two things for each [...]
I'm not sure I got this right, but the reason for having value buffers handled by user code, is that I don't want any dynamic memory allocation on the objects. I still wanted the user to have control on (statically) allocated memory sizes.

rovo89, 2020-12-22:
Here's a start of an implementation: https://gist.github.com/rovo89/d3d1187b802295bb4e0dc612be6c4f87
More like a PoC, it doesn't cover all my ideas and obviously doesn't implement the IotWebConf stuff.

I believe that the value buffers are part of the parameter object and should therefore be statically allocated, right? The `valueRef()` function still provides direct access to the variable and as it returns a reference, it can be assigned to. Not fully thought yet though.

Do you have a timeline for the v3.0.0 release? I could experiment with this more, but with the holidays and family time ahead, not sure how much free time I'll have in the next 1-2 weeks.

prampec, 2020-12-22:
My idea is to release v3.0.0 in the very beginning of January 2021.

prampec, 2020-12-22:
Btw. the HackingGuide is published.

prampec, 2021-04-14:
While some of these ideas are already implemented (thank you again), I keep this issue open, as some of the ideas still need consideration.