prampec/IotWebConf #157: Missing / Old files when installing through PlatformIO in VS Code

bedawi, 2020-12-22:
Hi Balázs Kelemen, many thanks for your work!

Unfortunately when installing your library (v2.3.3) using PlatformIO in VS Code, some files from your repository are missing and old files (like IotWebConfCompatibility.*) are installed. This leads to some problems...

prampec, 2020-12-22:
Master repository does not contain v2.X.X any more. It is v3.0.0. You can either migrate your code according to the migration guide (recommended), or use branch v2.x.x, thus ```https://github.com/prampec/IotWebConf.git#v2.x.x```

prampec, 2020-12-22:
(Master repository version does not automatically reflects the published release version. The published release version is still v2.3.3 for the time being.)

bedawi, 2020-12-22:
Alright, thanks. The issue here is that the examples shown in PlatformIO are already showing code that needs a newer version.

closed 2020-12-27 by prampec