prampec/IotWebConf #159: distorted characters when loading unset parameters

shmick, 2020-12-27:
I'm using v3.0 of the library and am just getting started with using parameters. When the unit reboots, I see a bunch of distorted characters in the input boxes. Is there a way to set some default null values?

Here's what the input boxes look like via HTML:

```
</fieldset><fieldset id='mqtt_conf'><legend>MQTT Configuration</legend><div class=''><label for='mqttUser'>MQTT User</label><input type='text' id='mqttUser' name='mqttUser' maxlength=64 placeholder='' value='ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿˆh!@' /><div class='em'></div></div>
  | <div class=''><label for='mqttPass'>MQTT Pass</label><input type='password' id='mqttPass' name='mqttPass' maxlength=64 placeholder='' value='' ondblclick="pw(this.id)"/><div class='em'></div></div>
  | <div class=''><label for='mqttHost'>MQTT Host</label><input type='text' id='mqttHost' name='mqttHost' maxlength=64 placeholder='' value='ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿˆh!@' /><div class='em'></div></div>
  | <div class=''><label for='mqttEnabledXParam'>MQTT Enabled</label><select id='mqttEnabledXParam' name='mqttEnabledXParam' />
  | <option value=''>Enabled</option>
```

```
#define STRING_LEN 64

char mqttUserValue[STRING_LEN];
char mqttPassValue[STRING_LEN];
char mqttHostValue[STRING_LEN];
char mqttEnabledValue[STRING_LEN];

static char mqttEnabledValues[][STRING_LEN] = {"enabled", "disabled"};
static char mqttEnabledNames[][STRING_LEN] = {"Enabled", "Disabled"};

IotWebConfParameterGroup mqttGroup = IotWebConfParameterGroup("mqtt_conf", "MQTT Configuration");
IotWebConfTextParameter mqttUserParam = IotWebConfTextParameter("MQTT User", "mqttUser", mqttUserValue, STRING_LEN);
IotWebConfPasswordParameter mqttPassParam = IotWebConfPasswordParameter("MQTT Pass", "mqttPass", mqttPassValue, STRING_LEN);
IotWebConfTextParameter mqttHostParam = IotWebConfTextParameter("MQTT Host", "mqttHost", mqttHostValue, STRING_LEN);
IotWebConfSelectParameter mqttEnabledParam = IotWebConfSelectParameter("MQTT Enabled", "mqttEnabledXParam", mqttEnabledValue, STRING_LEN, (char *)mqttEnabledValues, (char *)mqttEnabledNames, sizeof(mqttEnabledValues) / STRING_LEN, STRING_LEN);

  mqttGroup.addItem(&mqttUserParam);
  mqttGroup.addItem(&mqttPassParam);
  mqttGroup.addItem(&mqttHostParam);
  mqttGroup.addItem(&mqttEnabledParam);

```

prampec, 2020-12-27:
The easiest way is to modify the config version, where it will not load anything from EEPROM on boot time, but use defaults.

shmick, 2020-12-27:
> The easiest way is to modify the config version, where it will not load anything from EEPROM on boot time, but use defaults.

Thanks. I didn't realize that the config version was used in this way and wasn't using a config version.

closed 2020-12-27 by shmick