prampec/IotWebConf #161: [V3.0.0] isChecked() broken

coelner, 2020-12-31:
https://github.com/prampec/IotWebConf/blob/8493178020766b65301490dc6be24ea15eaa2a9d/examples/IotWebConf03CustomParameters/IotWebConf03CustomParameters.ino#L139

I tried to adapt the checkbox implementation in my code. The debug output:
```
iwcThingName: NTP-Clock-RGBLED
iwcApPassword: <was not changed>
iwcWifiSsid: WLAN 2.4GHz
iwcWifiPassword: <was not changed>
iwcApTimeout: 30
ntpServer: pool.ntp.org
max brightness: 200
allDotsOn: selected
followingHour : selected
Config version: dem6
Config size: 421
Saving configuration
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'NTP-Clock-RGBLED'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'WLAN-2.4GHz'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   |-- [Time]
|   |   \-- 'ntpServer' with value: 'pool.ntp.org'
|   \-- [LED]
|       |-- 'max brightness' with value: '200'
|       |-- 'singleSecond' with value: 'selected'
|       |-- 'allDotsOn' with value: 'selected'
|       \-- 'followingHour ' with value: 'selected'
\-- [hidden]
```
https://github.com/prampec/IotWebConf/blob/2da338e2aa2d67d9642d4c78ae89819c9a483829/src/IotWebConfParameter.h#L296

I assume that the form post is correctly not returning the singleSecond char array, but however this ends in the wrong result.

rexfue, 2021-01-01:
Same here: you can switch from 'NOT selected' to 'selected' but not back.

EvTheFuture, 2021-01-02:
Same here.

Can someone explain why checkbox is inheriting from TextParamter instead of say NumverParamter which uses less space?

Can't we just store a 16 bit int (or maybe 8 bits) instead of a complete string?

EvTheFuture, 2021-01-02:
I found the problem and have created Pull Request #162

prampec, 2021-01-03:
Thank you for the feedback! Problem was fixed in the main branch. Please test it, and in case of any trouble reopen this issue!

closed 2021-01-03 by prampec