prampec/IotWebConf #166: Implementation in the header file and default password length

4m1g0, 2021-01-19:
Hello, 

First of all I would like to congratulate the developers, specially @prampec for the work done in the last months, the improvements made to the proyect are really exciting. 

After going through the new code I have 2 questions/sugestions to discuss:

1. The implementation of the `IoTWebConf2ESP32HTTPUpdateServer` is done in the header file, this makes it to fail compilation for me, as we have the `IotWebCong.h` included in several files of the project:

```
In function `HTTPUpdateServer::HTTPUpdateServer(bool)':
multiple definition of `HTTPUpdateServer::HTTPUpdateServer(bool)'
first defined here
```
Is there a reason why it was made that way? I tried moving the implementation to a cpp file and work fine: https://github.com/prampec/IotWebConf/commit/a90d54f149fbbe3397c5a03b7d161538199401d1

2. The standard wifi password has a length of 8-64 bytes but the current default wifi password length is 32 bytes, this makes it mandatory for every project to change the default length to 65 to support all users. In our project we ran into this problem when some users complained that they couldn't fit their passwords into the wifi password field.

I know this was discussed before, but with the new implementation of the parameters the buffers of each parameter are handled in a more clean way than in the previos version and there is no problem with dynamic buffer allocation anymore as discussed here: https://github.com/prampec/IotWebConf/pull/94

I think it would be good for this project, to set the wifi password length to the standard 65 bytes by default to make this more user friendly.

My suggestion is something like this: https://github.com/prampec/IotWebConf/commit/8f3497e33636b97188fccbee46f109b7cd8641e2

Tell me if you want me to make any modifications and/or pull requests.

Thank you!!

prampec, 2021-01-20:
About the header - I'm not sure what the problem is with that. I all C++ template things are implemented in header files. E.g. [ESP8266WebServer-impl.h](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h)
Maybe a [pramga once](https://en.wikipedia.org/wiki/Pragma_once) should be used?

About the password - Passwords are in IotWebConf core means WifiPasswords, so there are no sense define a IOTWEBCONF_WIFI_PASSWORD_LEN, IOTWEBCONF_PASSWORD_LEN means that now. You are free to redefine this value with the value of your choice. See [Hacking guide](https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#compile-time-configuration) for changing compile time configuration.

4m1g0, 2021-01-20:
IOTWEBCONF_PASSWORD_LEN is used in the ap password (_apPasswordParameter) and the Wifi password (wifiPasswordParameter), so if you change that define, you are changing both, that's the reason I added a different define, but changing both buffers would work too...

In any case, this was only a suggestion. in my opinion the library should use the standard 64bytes by default in order to support all wifi passwords without any changes. If some project for some reason wants to limit the password size can do it as you mention.

Regarding the header. You are right. I'll further investigate it and comment about it soon. I also noticed that the ESP32 framework includes the HTTPUpdateServer in the next release: https://github.com/espressif/arduino-esp32/commit/3054bdf5a5d4d9fe79d36d0dd1b287f5a70109f3

Thank you!

4m1g0, 2021-01-21:
Hello again,

Regarding the implementation inside the header file... I've made many tests and I have some things clear now:

There are 3 ways of declaring a method of a class:

1. In the cpp file
2. In the header file, inside the curly brazes of the class 
3. In the header file, outside the curly brazes of the class

With 1, the code is compiled once and the symbols are used where is needed. With 2 the code is included in every place where it's used but it's marked as inline by the compiler so it's legal to re-use the same code again. And with 3, the code is included in every place but it's not marked as inline by the compiler so the linker finds it twice and considers it illegal. (Here, this is explained quite well: https://stackoverflow.com/questions/1809679/difference-between-implementing-a-class-inside-a-h-file-or-in-a-cpp-file )

Now the question is: Why does it work in the example you showed? https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h

The answer is exactly because they use templates... If the class has a template, it is treated differently and that becomes legal again. I would really love to understand what happens under the hood there, but this is going beyond my knowledage, and couldn't find the answer yet. But I think it might be related to the fact that templates are not actual implementations and cannot be fully compiled until declared, so it makes sense that they are treated differently by the compiler.

To prove this I created a very simple project with only 2 commits: https://github.com/4m1g0/test_error/commits/master

The first commit emulates the stucture of the current IotWebConf2ESP32HTTPUpdateServer class and it fails when linking: 
```
Building in release mode
Compiling .pio/build/heltec_wifi_lora_32/src/ClassThatUsesTheH.cpp.o
Compiling .pio/build/heltec_wifi_lora_32/src/main.cpp.o
Linking .pio/build/heltec_wifi_lora_32/firmware.elf
.pio/build/heltec_wifi_lora_32/src/main.cpp.o: In function `ClassWithImplementation::ClassWithImplementation(int)':
/home/oscar/Documents/PlatformIO/Projects/test_reimplementation_bare/src/ClassWithImplementation.h:12: multiple definition of `ClassWithImplementation::ClassWithImplementation(int)'
.pio/build/heltec_wifi_lora_32/src/ClassThatUsesTheH.cpp.o:/home/oscar/Documents/PlatformIO/Projects/test_reimplementation_bare/src/ClassWithImplementation.h:12: first defined here
.pio/build/heltec_wifi_lora_32/src/main.cpp.o: In function `ClassWithImplementation::ClassWithImplementation(int)':
main.cpp:(.text._ZN23ClassWithImplementationC2Ei+0x0): multiple definition of `ClassWithImplementation::ClassWithImplementation(int)'
.pio/build/heltec_wifi_lora_32/src/ClassThatUsesTheH.cpp.o:ClassThatUsesTheH.cpp:(.text._ZN23ClassWithImplementationC2Ei+0x0): first defined here
collect2: error: ld returned 1 exit status
*** [.pio/build/heltec_wifi_lora_32/firmware.elf] Error 1
```

The second commit only adds a template to the same class:
```Building in release mode
Compiling .pio/build/heltec_wifi_lora_32/src/ClassThatUsesTheH.cpp.o
Compiling .pio/build/heltec_wifi_lora_32/src/main.cpp.o
Linking .pio/build/heltec_wifi_lora_32/firmware.elf
Retrieving maximum program size .pio/build/heltec_wifi_lora_32/firmware.elf
Checking size .pio/build/heltec_wifi_lora_32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.5% (used 14708 bytes from 327680 bytes)
Flash: [==        ]  16.0% (used 209827 bytes from 1310720 bytes)
```

There are 2 options to fix this:
* Move the implementation to a cpp file
* Move the methods inside the curly braces of the class

prampec, 2021-03-28:
About the standard password length: You are right, that it would be much nicer to follow the standards. On the other hand I'm afraid we are beyond the point where we can change values, that will effects saved configuration of live devices.
With v3.0.0, a typed-parameter approach is introduced, where you are free to define storage size for any type of parameter. Maybe one day this will be the default for the system parameters as well, and then we can apply the standards.

prampec, 2021-04-14:
The strange thing with this problem is that I can compile for all the platforms without any issue.
Still I will address this issue.

4m1g0, 2021-04-14:
I'm using platformio on vscode with Espressif 32@3.1.0

prampec, 2021-04-14:
I just have found a pull request about this. That is all we need :)
https://github.com/prampec/IotWebConf/pull/167

prampec, 2021-04-17:
I just have applied pull request #167 containing this fix, but I'm not happy with the result.
Now all Esp32 projects needs Update library to IotWebConf be compiled, even if the WebUpdate is not used at all. E.g. everyone must add "Update" to libdeps in platformIO.

Is there a way we can avoid that? I really don't like the way it is now.

4m1g0, 2021-04-18:
It definitely shouldn't be that way. Update.h is a core library and it's included in the Espressif framework however looks like due to the way libraries are compiled those libraries are not available for other libraries. If you import Update.h in your user code, it automatically works without installing any additional library. New things I learn...

This problem leaves us with only the other option I mentioned earlier which is including the code in the header file but inside the curly braces... I think this fixes both problems: #183 I've tested it in platformio with ESP32.

prampec, 2021-06-12:
Thank you for fixing this in  #183 (Commit: 4dc0c79560d103288a6f5f1f11140fb22e3800e5)

closed 2021-06-12 by prampec