prampec/IotWebConf #169: Assigning an initial value to variables in version 3.0.0

PilotIoTSystems, 2021-02-11:
Testing was performed on the D1 mini (ESP8266) module using the IotWebConf-3.0.0 Examples "IotWebConf03CustomParameters" Sketch. In the previous version (2.3.3) the following assignment worked:

char stringParamValue[STRING_LEN] = "Something";

As a result, the variable "stringParamValue" took on this value, and this initial value also appeared on the config portal. However, this does not work now, the value of the variable "stringParamValue" remains empty (""). I'm afraid something I'm thinking wrong. Please help with how to enter an initial value in sketch in the new version, which can then be changed in the setup portal if needed. Thanks in advance

PilotIoTSystems, 2021-02-11:
Looks like I managed to find a solution! The solution is as follows:

char stringParamValue[STRING_LEN] = "Something";

IotWebConf-3.0.0 Examples Line in "IotWebConf03CustomParameters" Sketch:

IotWebConfTextParameter stringParam = IotWebConfTextParameter("String param", "stringParam", stringParamValue, STRING_LEN);

With my addition, which determines the initial value:

IotWebConfTextParameter stringParam = IotWebConfTextParameter("String param", "stringParam", stringParamValue, STRING_LEN, stringParamValue);

So the initial value is OK too!

prampec, 2021-02-11:
You must provide a default value in parameter creation to have an initial value, to be used in case no configuration ever made on your config portal in the device before.

closed 2021-02-11 by prampec