prampec/IotWebConf #170: Error in Example no matching function for call in iotWebConf.setFormValidator(&formValidator)

Agrumbach, 2021-02-15:
If i compile the example MQTTRelay in PlatformIO for Board wemos_d1_mini32 i get this Error:

src\main.cpp: In function 'void setup()':
src\main.cpp:109:45: error: no matching function for call to 'iotwebconf::IotWebConf::setFormValidator(bool (*)(iotwebconf::WebRequestWrapper*))'
   iotWebConf.setFormValidator(&formValidator);
                                             ^
In file included from src\main.cpp:40:0:
.pio\libdeps\wemos_d1_mini32\IotWebConf\src/IotWebConf.h:298:8: note: candidate: void iotwebconf::IotWebConf::setFormValidator(std::function<bool()>)
Compiling .pio\build\wemos_d1_mini32\lib94d\WebServer\Parsing.cpp.o
   void setFormValidator(std::function<bool()> func);
        ^
.pio\libdeps\wemos_d1_mini32\IotWebConf\src/IotWebConf.h:298:8: note:   no known conversion for argument 1 from 'bool (*)(iotwebconf::WebRequestWrapper*)' to 'std::function<bool()>'
*** [.pio\build\wemos_d1_mini32\src\main.cpp.o] Error 1

arvdsar, 2021-02-22:
I stumbled upon the same error for Wemos D1 (esp8266). Code was running fine with 3.0.0 of 20-12-2020. I deleted the library and replaced with 3.0.0pre2 and now I receive same error. 
Rolled back by deleting library and added 3.0.0 again using prampec/IotWebConf@^3.0.0 in platformio.ini and the code is running fine again.

(adding 3.0.0pre2 using platformio.ini does not work by the way!)

arvdsar, 2021-02-22:
Hi,
I solved my issue which probably helps you too!
search your code for:
boolean formValidator();

replace with:
bool formValidator(iotwebconf::WebRequestWrapper* webRequestWrapper);

I expect you find this twice in your code. I noticed this when comparing the original examples on which my code was based and compared it with the most recent example.

Agrumbach, 2021-02-22:
Hi,
thanks for your help but it dosent work for me. I have reinstalled the lib prampec/IotWebConf@^3.0.0 and check the code. The function hase the format formValidator(iotwebconf::WebRequestWrapper* webRequestWrapper); the differnt ist that i use the D1 ESP32 i will check the ESP8266 and check with this board.

arvdsar, 2021-02-22:
The 3.0.0 uses: boolean formValidator();
I downloaded the library code manually and put this in lib folder (and deleted the one from libdeps while commenting the prampec/IotWebConf@^3.0.0 in platform.ini (it won't get the 3.0.0pre2 version in my case. I keeps the old code in libdeps). So maybe you are using correct example but old library in platformio.

Agrumbach, 2021-02-22:
You rigth i create an new Project and then i have the version 3.0.0pre2 show in the libary.properties file in the IoTWebConf folder under the libdeps and it compiles without errors.

Thank You for your help!

thelastWallE, 2021-02-26:
I get this error if i use the example:
`no suitable constructor exists to convert from "bool (*)(iotwebconf::WebRequestWrapper *webRequestWrapper)" to "std::function<bool ()>"`

prampec, 2021-03-28:
Please upgrade to v3.0.0 to clear all version-mismatch issues.

closed 2021-03-28 by prampec