prampec/IotWebConf #171: Static IP or DHCP

fusionstream, 2021-02-17:
It seems no one has asked this before.

Could do with an option for static IP or DHCP. Would be nice to be part of the default options, otherwise, you have allow the WiFi.begin call in connectWifi to be overloaded with static IP details.

Use case, I have sensor nodes that I want to set up easily since I will have a ton of them and not all sensor types are the same (so I use custom parameters). This node will take a reading then go into a deepSleep. As power usage is important, I want to set a static ip as it dramatically reduces the on time. However, I initially will need dhcp support so i don't have to worry about assignments till much later in the installation.

fusionstream, 2021-02-17:
Found a sample bit of code in Conf09 that might support it without modifying the library manually. Will post back when I have working code. But the point still stands, having it as part of the native interface would be great!

PilotIoTSystems, 2021-06-07:
Optional DHCP and Static IP would be a really good solution! How is this development?

prampec, 2021-06-08:
It is doable, as shown in example IotWebConf09CustomConnection.
I keep this thread open to provide a more complete example with all the GUI items.

PilotIoTSystems, 2021-06-10:
I have compiled a version that meets the wishes of a **fusionstream** in which the Static IP or DHCP check box can be selected. It works for me...

/**
 * IotWebConf09CustomConnection.ino -- IotWebConf is an ESP8266/ESP32
 *   non blocking WiFi/AP web configuration library for Arduino.
 *   https://github.com/prampec/IotWebConf 
 *
 * Copyright (C) 2020 Balazs Kelemen <prampec+arduino@gmail.com>
 *
 * This software may be modified and distributed under the terms
 * of the MIT license.  See the LICENSE file for details.
 */

/**
 * Example: Custom connection
 * Description:
 *   This example is for advanced users only!
 *   In this example custom connection handler methods are defined
 *   to override the default connecting behavior.
 *   Also, three custom parameters are introduced, that are used
 *   at the connection.
 *   (See previous examples for more details!)
 * 
 * Hardware setup for this example:
 *   - An LED is attached to LED_BUILTIN pin with setup On=LOW.
 *   - [Optional] A push button is attached to pin D2, the other leg of the
 *     button should be attached to GND.
 */

#include <IotWebConf.h>
#include <IotWebConfUsing.h> // This loads aliases for easier class names.

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "testThing";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "smrtTHNG8266";

#define STRING_LEN 128
#define NUMBER_LEN 32

// -- Configuration specific key. The value should be modified if config structure was changed.
#define CONFIG_VERSION "dem9_reworked"

// -- When CONFIG_PIN is pulled to ground on startup, the Thing will use the initial
//      password to buld an AP. (E.g. in case of lost password)
#define CONFIG_PIN 2

// -- Status indicator pin.
//      First it will light up (kept LOW), on Wifi connection it will blink,
//      when connected to the Wifi it will turn off (kept HIGH).
#define STATUS_PIN LED_BUILTIN

// -- Method declarations.
void handleRoot();
bool connectAp(const char* apName, const char* password);
void connectWifi(const char* ssid, const char* password);
// -- Callback methods.
void configSaved();
bool formValidator(iotwebconf::WebRequestWrapper* webRequestWrapper);

DNSServer dnsServer;
WebServer server(80);

char ipAddressValue[STRING_LEN];
char gatewayValue[STRING_LEN];
char netmaskValue[STRING_LEN];
char StaticIP_enabledVal[STRING_LEN];
boolean StaticIP_enabled = false;

[... 123 lines cut ...]
  // -- Custom AP settings
  return WiFi.softAP(apName, password, 4);
}
void connectWifi(const char* ssid, const char* password)
{
if (StaticIP_enabled) {
  ipAddress.fromString(String(ipAddressValue));
  netmask.fromString(String(netmaskValue));
  gateway.fromString(String(gatewayValue));

  
  //WiFi.config(ipAddress, gateway, netmask);

  if (!WiFi.config(ipAddress, gateway, netmask)) {
    Serial.println("STA Failed to configure");
  }
  Serial.print("ip: ");
  Serial.println(ipAddress);
  Serial.print("gw: ");
  Serial.println(gateway);
  Serial.print("net: ");
  Serial.println(netmask);
}
  WiFi.begin(ssid, password);
}

PilotIoTSystems, 2021-06-10:
`
[image]
`

PilotIoTSystems, 2021-06-10:
`/**
 * IotWebConf09CustomConnection.ino -- IotWebConf is an ESP8266/ESP32
 *   non blocking WiFi/AP web configuration library for Arduino.
 *   https://github.com/prampec/IotWebConf 
 *
 * Copyright (C) 2020 Balazs Kelemen <prampec+arduino@gmail.com>
 *
 * This software may be modified and distributed under the terms
 * of the MIT license.  See the LICENSE file for details.
 */

/**
 * Example: Custom connection
 * Description:
 *   This example is for advanced users only!
 *   In this example custom connection handler methods are defined
 *   to override the default connecting behavior.
 *   Also, three custom parameters are introduced, that are used
 *   at the connection.
 *   (See previous examples for more details!)
 * 
 * Hardware setup for this example:
 *   - An LED is attached to LED_BUILTIN pin with setup On=LOW.
 *   - [Optional] A push button is attached to pin D2, the other leg of the
 *     button should be attached to GND.
 */

#include <IotWebConf.h>
#include <IotWebConfUsing.h> // This loads aliases for easier class names.

// -- Initial name of the Thing. Used e.g. as SSID of the own Access Point.
const char thingName[] = "testThing";

// -- Initial password to connect to the Thing, when it creates an own Access Point.
const char wifiInitialApPassword[] = "smrtTHNG8266";

#define STRING_LEN 128
#define NUMBER_LEN 32

// -- Configuration specific key. The value should be modified if config structure was changed.
#define CONFIG_VERSION "dem9_reworked"

// -- When CONFIG_PIN is pulled to ground on startup, the Thing will use the initial
//      password to buld an AP. (E.g. in case of lost password)
#define CONFIG_PIN 2

// -- Status indicator pin.
//      First it will light up (kept LOW), on Wifi connection it will blink,
//      when connected to the Wifi it will turn off (kept HIGH).
#define STATUS_PIN LED_BUILTIN

// -- Method declarations.
void handleRoot();
bool connectAp(const char* apName, const char* password);
void connectWifi(const char* ssid, const char* password);
// -- Callback methods.
void configSaved();
bool formValidator(iotwebconf::WebRequestWrapper* webRequestWrapper);

DNSServer dnsServer;
WebServer server(80);

char ipAddressValue[STRING_LEN];
char gatewayValue[STRING_LEN];
char netmaskValue[STRING_LEN];
char StaticIP_enabledVal[STRING_LEN];
boolean StaticIP_enabled = false;

IotWebConf iotWebConf(thingName, &dnsServer, &server, wifiInitialApPassword, CONFIG_VERSION);
// -- You can also use namespace formats e.g.: iotwebconf::ParameterGroup
[... 121 lines cut ...]
  // -- Custom AP settings
  return WiFi.softAP(apName, password, 4);
}
void connectWifi(const char* ssid, const char* password)
{
if (StaticIP_enabled) {
  ipAddress.fromString(String(ipAddressValue));
  netmask.fromString(String(netmaskValue));
  gateway.fromString(String(gatewayValue));

  
  //WiFi.config(ipAddress, gateway, netmask);

  if (!WiFi.config(ipAddress, gateway, netmask)) {
    Serial.println("STA Failed to configure");
  }
  Serial.print("ip: ");
  Serial.println(ipAddress);
  Serial.print("gw: ");
  Serial.println(gateway);
  Serial.print("net: ");
  Serial.println(netmask);
}
  WiFi.begin(ssid, password);
}`

prampec, 2021-06-10:
Sounds great, thanks. Would be nice to set up some javascipt dynamism on OptionalParameterGroup.

puyiot, 2021-06-12:
I do that too.
[image]