prampec/IotWebConf #172: Beginner's miscellaneous issues (Releases, clear credentials, configuration page)

gnbl, 2021-02-24:
Trying https://github.com/prampec/IotWebConf/blob/v2.3.3/examples/IotWebConf01Minimal/IotWebConf01Minimal.ino with ESP32 I came across a number of issues.

- releases contradict readme
> Lately version 3.0.0 is released. 
vs. 
https://github.com/prampec/IotWebConf/releases/latest -> https://github.com/prampec/IotWebConf/releases/tag/v2.3.3

- configuration 
  - an error (ap password too short) on the configuration page clears the wifi password - tedious
  - not entering a new ap password on initial configuration causes intermediate state
  - how to clear the configuration? 
    - there should be a public method to do this without having to use hardware via `setConfigPin` - is there?
    - regardless of providing a new thingName and initial Password and passing different version string, old data is loaded
    - `iotWebConf.resetWifiAuthInfo();` did not seem to have effect
  - accessing config page requires login as "admin" - if there's only one possible user, could this be preset and have a password hint? or do not use HTTP auth but handle session instead.
  - HTML errors in "Config ESP":
    - extraneous closing tag `<form action='' method='post'><input type='hidden' name='iotSave' value='true'>
'><legend>System configuration</legend>`
  - broken fieldset `<div class='em'></div></div> </fieldset><button type='submit'>Apply</button></form>`

I'm happy that I managed to reset/overwrite the network credentials somehow and I'm now on to add custom properties:-)

prampec, 2021-02-24:
Huhh! Lot of questions.
Let's call v3.0.0 a pre-release, I know, this is confusing, I hope final version will be released very soon.

> an error (ap password too short) on the configuration page clears the wifi password - tedious

It is overall the best practice not to send any password. IotWebConf utilizes a very simple HTML interface to reduce memory footprint and to avoid any files to be loaded into flash. Simple HTML post will reload the actual page, where I do not want to send back the password to the browser. (But you are free to utilize 3rd party javascript framework rendering the web page with async post.

> not entering a new ap password on initial configuration causes intermediate state

I'm not sure what you mean. Without an AP password the Thing will remain in AP mode. Please provide more details on this!

> how to clear the configuration?
>        there should be a public method to do this without having to use hardware via setConfigPin - is there?

That sounds reasonable.

>        regardless of providing a new thingName and initial Password and passing different version string, old data is loaded

This must not happen.

>        iotWebConf.resetWifiAuthInfo(); did not seem to have effect

Please read documentation on resetWifiAuthInfo! Also check the MultipleWifiAddition used by the IotWebConf15MultipleWifi example in the topic!

>    accessing config page requires login as "admin" - if there's only one possible user, could this be preset and have a password hint? or do not use HTTP auth but handle session instead.

I believe you cannot preset user name in HTTP auth like this. More sophisticated login is just not worth to implement as mentioned above, where I talk about simple HTTP solution.

>    HTML errors in "Config ESP":

Please confirm whether it is the v3.0.0-pre.2 version, where you have found these!

gnbl, 2021-02-25:
Thanks for your response and sorry about the "dump". Time is scarce, but I'll try to reproduce and provide better info when I get a chance.

prampec, 2021-02-25:
I'm happy to get any response, so thank You for your feedback.

prampec, 2021-04-14:
> how to clear the configuration?

There is a method for any ConfigItem named applyDefaultValue(), that will reset the value to defaults.
Now, using this for e.g. iotWebConf.getSystemParameterGroup() will reset all parameters in the System group.

This function will be available in 3.0.2 because in previous version this method was not public.

(For the the other issues please open separate tickets!)

closed 2021-04-14 by prampec