prampec/IotWebConf #173: Example IotWebConf04UpdateServer showing error

worthapenny, 2021-03-07:
Hi there,

First of all thanks for sharing the work!

I copied the Example (UpdateServer) into PlatformIO and it is showing some error:

[image]

[image]

Error causing lines (which is hard to see in the screenshot):
```CPP
  // -- Define how to handle updateServer calls.
  iotWebConf.setupUpdateServer(
    [](const char* updatePath) { httpUpdater.setup(&server, updatePath); },
    [](const char* userName, char* password) { httpUpdater.updateCredentials(userName, password); });
```

My limited knowledge is not helping me to troubleshoot this error. Hence, any suggestion would be appreciated.

prampec, 2021-03-28:
You have some mix-up in your library/examples version. Please make sure you are using IotWebConf 3.0.0-pre.X version (or later) on both library and examples side.

prampec, 2021-04-14:
Please update to IotWebConf 3.0.1 as there was some mix-up with naming 3.0.0*

closed 2021-04-14 by prampec