prampec/IotWebConf #174: setting hostname issue

societyofrobots, 2021-03-08:
Had trouble setting the hostname; it was only displaying 'espressif'. This caused an issue when having more than one ESP32 connected on the same router.

[Others also had the problem, but you weren't able to solve it at the time.](https://github.com/prampec/IotWebConf/issues?q=is%3Aissue+hostname)

I managed to solve it with a dirty hack (I'm a weak programmer, forgive me!).

In setup(), right after 
  server.onNotFound([](){ iotWebConf.handleNotFound(); });
I added this code:
  
```
WiFi.disconnect(true);
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE); // call is only a workaround for bug in WiFi class
WiFi.setHostname(thingName.c_str());
```

This is the info that helped me do this:

https://stackoverflow.com/questions/57570586/how-to-correctly-set-the-hostname-for-the-tcp-ip-adapter-on-the-esp32
https://github.com/espressif/arduino-esp32/issues/806#issuecomment-589621154
https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-589619286

prampec, 2021-03-28:
Yes. I know, that there are some tricks for ESP8266 and other tricks for ESP32.
E.g. some extra delays in some particular places might also has effect on the hostname settings. It would be nice to have these thing solved by the libraries.
Apparently I set hostname and MDNS before before connecting to WiFi. Should I do if afterwards?

societyofrobots, 2021-03-28:
I just tried it before iotWebConf.init(); and it did not work. Only works after.

prampec, 2021-03-29:
Can you please try and add following line to checkWifiConnection method before ```return true``` statement (line: 806)?!
```
  WiFi.setHostname(this->_thingName);
```

societyofrobots, 2021-03-29:
Tried, nope, doesn't work.

EricDuminil, 2022-02-21:
After a bit of googling and experimenting with ESP8266 and ESP32, I found out that:

* `WiFi.getHostname()` and `WiFi.setHostname()` have both been defined for ESP8266 and ESP32, so there's no need for #ifdefs 
* `WiFi.setHostname()` needs to happen right before `WiFi.begin()` on ESP8266. Otherwise, the hostname gets reset to `ESP-xxxxxx`, e.g. as soon as `WiFi.mode(WIFI_STA)` is called.
* `WiFi.getHostname()` will only return the set hostname after `WiFi.begin()` on ESP32.

As far as I can tell, those changes ( https://github.com/prampec/IotWebConf/compare/master...EricDuminil:fix/set_hostname?expand=1 ) allow to reliably set the hostname on ESP8266 and ESP32. @prampec I'll prepare a pull-request if you're interested.

Sadly, it doesn't mean that mDNS will work reliably. This technology apparently depends on the client, the router, and moon phases.

PR #241 (2022-04-20): Fix/set hostname
Should fix #174 .
closed 2022-04-20 by prampec