prampec/IotWebConf #176: Reset password + parameters to default in software, without using ConfigPin

societyofrobots, 2021-04-10:
The physical unit is in an isolated location that users cannot trigger a physical pin. I had set up a [software based reset for the password](https://github.com/prampec/IotWebConf/issues/130) and other various parameters in v2 that worked great.

Now that I'm converting to v3.0, the reset to default code is no longer working.

I need to reset wifi password, wifi name, and the new v3.0 parameters via software to the default values (thingName.c_str(), wifiInitialApPassword, intParam, CheckboxTParameter).

How might I best do this?

prampec, 2021-04-11:
All ConfigItem implements applyDefaultValue() method, that restores hardcoded default value.
If you want to reset e.g. SytemParameters, you can do it by IotWebConf.getSystemParameterGroup().applyDefaultValue().
And don't forget to save configuration with saveConfig() !

societyofrobots, 2021-04-12:
Forgive me, I'm not the strongest of programmers. I tried this:
```
IotWebConf.getSystemParameterGroup().applyDefaultValue();
//IotWebConf.intBLEParam.applyDefaultValue();
iotWebConf.saveConfig();
```

But get this compile error:
```
pass_reset.h:31:11: error: expected unqualified-id before '.' token
 IotWebConf.getSystemParameterGroup().applyDefaultValue();
           ^
exit status 1
expected unqualified-id before '.' token
```

prampec, 2021-04-12:
Small 'i' and '->' instead of some dots, thus:
```iotWebConf.getSystemParameterGroup()->applyDefaultValue()```

This will reset the thing name also. So you might want to go with:
```iotWebConf.getApPasswordParameter()->->applyDefaultValue()```
as seen in the Hacking guide: https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#accessing-system-properties

I hope no syntax error this time. :)
Meanwhile please learn the difference between object-class, and object-instance! ;) This is crucial.

societyofrobots, 2021-04-13:
I had actually previously tried lower case i but it gave more errors.

this code:
`iotWebConf.getApPasswordParameter()->->applyDefaultValue();`

gives me this error:
```
error: expected unqualified-id before '->' token
 iotWebConf.getApPasswordParameter()->->applyDefaultValue();
                                      ^
exit status 1
expected unqualified-id before '->' token
```

this code:
`iotWebConf.getSystemParameterGroup()->applyDefaultValue();`

gives this error:
```
error: 'virtual void iotwebconf::ParameterGroup::applyDefaultValue()' is protected
   void applyDefaultValue() override;
        ^
In file included from C:\...blah.ino:115:0:
pass_reset.h:33:57: error: within this context
 iotWebConf.getSystemParameterGroup()->applyDefaultValue();
                                                         ^
exit status 1
within this context
```

prampec, 2021-04-13:
"I hope no syntax error this time. :)" - Still had one: ```->->``` is certainly not correct.
Correct form ```iotWebConf.getApPasswordParameter()->applyDefaultValue()```

Still applyDefaultValue() shouldn't be protected. This should be addressed.

societyofrobots, 2021-04-13:
> Still applyDefaultValue() shouldn't be protected. This should be addressed.

I manually modified the library files, moving every instance of applyDefaultValue I could find from Protected into Public.

Not only does:
`iotWebConf.getSystemParameterGroup()->applyDefaultValue();`
force default values to system parameters, it also does it to the custom parameters. Perfect, because that's what I wanted.

Solved!

ps - Will 3.0.2 include this change?

closed 2021-04-13 by societyofrobots

prampec, 2021-04-14:
> Will 3.0.2 include this change?
Yes.

EricDuminil, 2022-02-13:
`iotWebConf.getSystemParameterGroup()->applyDefaultValue();` does not reset custom or hidden parameters, though, right?

I think that in order to reset everything, this call is required:

```
iotWebConf.getRootParameterGroup()->applyDefaultValue();
iotWebConf.saveConfig();
```