prampec/IotWebConf #177: username and the password for the login after config the WIFI-client?

Bergloewe007, 2021-04-13:
How can I find the password for the login after config the WIFI-client?

prampec, 2021-04-14:
Please check https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#accessing-system-properties

closed 2021-04-14 by prampec

Bergloewe007, 2021-04-14:
Thanks!
But, I come there not clear. I don't be a super C++-expert. I'm sorry.
I can't find this row in the code wheare I can see, how is the login UserName and login UserPassWord. Because I can't start over the config in the WIFI-Client.
In the HackingGuide I can see the names if the functions with this pointed to the paramters. And in the IotWebConf.h I can see the "&thing"-pointers to this. But no where are the parameters himself. In the IotWebConf.cpp file I don't find this also.

prampec, 2021-04-14:
Ohh, you might mean the Web-password? The web-login username is admin and the password is the same as the configured AP password.

If you are interested in the configured AP password to be read from the code, then...
We have Parameters in IotWebConf. AP password turns out to be a Parameter as well. You can get the AP password parameter by iotWebConf.getApPasswordParameter().
This will returns a Parameter pointer. On the parameter pointer you can access some fields directly e.g. apPasswordParameterPointer->valueBuffer

Now the valueBuffer is a character array containing the string value.