prampec/IotWebConf #179: is this can replace the connection part when connection with arduino?

youxiaojie, 2021-04-13:
I modified an old arduino project the original data collected by serial,now I hope to set data to internet. arudino with 8266 by serial, start from the demo, the wifi ssid and passwd is written in code, only connected with wifi in home. is this project allow me to select wifi connection, after connectted to wifi, I can write arduino corrected data code to serial and to internet?

prampec, 2021-04-14:
> is this project allow me to select wifi connection, after connectted to wifi

Currently the only way to select an alternative WiFi connection is when the primary network is not available, so secondary connection is tried automatically. There are ways to modify properties of the primary connection on the go, but this requires some advanced programming skills, where you have to be certain about the things you do. 

>  I can write arduino corrected data code to serial and to internet?

You can write a code to the ESP8266, to read serial input (RX, that is connected to the TX of a regular Arduino board) and send collected data over WiFi. IotWebConf helps you set up and manage the WiFi connections in case you don't want it to be hardcoded.

If I'm not answered the question you wanted to ask, please fill free to reopen this issue!

closed 2021-04-14 by prampec

youxiaojie, 2021-04-14:
serial port I am clear. thanks very much. I am a newbie to wifi:)
when I move between home and office, there is different ssid and passwd. 
1.you mean we can save these ssid/passwd in esp8266, they will auto selected, right?
2.if move to a third place, 8266 connect to wifi failed, I can add this new ssid/passwd when start?
3.if the same place has two ssid/passwd, they will choose one? and can not be changed if this connection works?
4.following upper question, if the wifi ap is power down, no signal, will it try next?
5.if the wifi ap is ok, but for some reason, it offline,will it try next?

thanks.

prampec, 2021-04-14:
I believe IotWebConf is capable for all you have listed here. Further more, if none of the configured WiFi networks are available it will fall back to AP mode for some seconds, so that you can make modifications on the configurations.
Please check IotWebConf15MultipleWifi.
(Just for clarification: You have used word "wifi ap" for existing WiFi networks, that is fine, I have understand that. But I always use AP for local access point create by the ESP8266 in context of IotWebConf.)

youxiaojie, 2021-04-16:
great!thanks!