prampec/IotWebConf #182: Bug Softap without encryption

Rotzbua, 2021-04-16:
## Environment

Chipset: eps32
Iotwebconf: 3.0.1
Compiler: PlatformIO 5.1.1

## Problem

Softap is unencrypted.

## Reproduce

I do not know how this exactly could happen.
I use custom configuration.

1. configured wifi & saved settings
2. configured custom fields & saved

Maybe I changed layout without updating the version number for eeprom storage.

## Expected behavior

If no valid password is available from config -> fallback to initial compiled password instead of open ap.

prampec, 2021-04-16:
You are right. This MUST be fixed ASAP.

prampec, 2021-04-16:
I cannot reproduce this. Please check the console:
Ready.
AP password was not set in configuration
State changing from: 0 to 1
Setting up AP: testThing

Rotzbua, 2021-04-16:
I took some code from the mqtt example. 
After I configured and reset the esp, there is an encrypted AP.
After another restart there is an open AP is started for the configured 30 seconds.
If there is no wifi where the esp could connect, it fallback again back to the unencrypted AP.
Really strange behavior 😕  .

I erased the complete flash of the esp32 and compiler cache and build the application again to avoid interference with old config.

The output is:
```
Config version: mqt2
Config size: 549
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'testThing'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: '1-ABCD-IOT'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
|   \-- [MQTT configuration]
|       |-- 'mqttServer' with value: ''
|       |-- 'mqttUser' with value: ''
|       \-- 'mqttPass' with value: <hidden>
\-- [hidden]
Ready.
State changing from: 0 to 2
Setting up AP: testThing
Use password: <hidden>
AP IP address: 192.168.4.1
AP timeout (ms): 30000
State changed from: 0 to 2
Connection to AP.
[E][WebServer.cpp:633] _handleRequest(): request handler not found
Request for connectivitycheck.gstatic.com redirected to 192.168.4.1

prampec, 2021-04-16:
Do you provide any password for the AP?
After "Apply" button you should see:
```
iwcThingName: testThing
iwcApPassword: <was not changed>
iwcWifiSsid: someSsid
iwcWifiPassword: <updated>
```

Rotzbua, 2021-04-16:
> Do you provide any password for the AP?
> After "Apply" button you should see:

Yes, I compiled my project with `IOTWEBCONF_DEBUG_PWD_TO_SERIAL` and it shows the correct password.

Rotzbua, 2021-04-16:
Ok, I found an issue, seems to be an framework bug: https://github.com/espressif/arduino-esp32/issues/5038
May keep this issue open as information for others until it is fixed in the framework.

prampec, 2021-04-16:
Thank you for pointing this out!

societyofrobots, 2021-04-19:
Confirming. I'm also seeing this serious security issue.

Unfortunately when I revert back to 1.0.4, I run into incompatibilities with other libraries I'm using.

Is there any quick fix that can be done via IotWebConf? How can we get the attention of the core developers?

Rotzbua, 2021-04-19:
> Is there any quick fix that can be done via IotWebConf? 

Force HTTP Auth like on lan access would be an solution. This would be vulnerable to sniffing but better than an non auth access to the conf.

prampec, 2021-04-19:
Can we somehow detect from the software whether we have an unprotected AP set up?

societyofrobots, 2021-04-19:
> Force HTTP Auth like on lan access would be an solution.

How could I do this? The sniffing vulnerability isn't an issue for me in the short term.

> Can we somehow detect from the software whether we have an unprotected AP set up?

I set to verbose and this is all it gave me when connecting by AP unsecured:
```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
```

I'd assume it had to be one of these changes in the last few months that broke it?
https://github.com/espressif/arduino-esp32/commits/master/libraries/WiFi/src

Rotzbua, 2021-04-19:
> Unfortunately when I revert back to 1.0.4, I run into incompatibilities with other libraries I'm using.

@societyofrobots You are using Arduino IDE right? Then you can try the development version `2.0.0-alpha1`. Just add the development url to the board manager:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

Rotzbua, 2021-04-27:
> 
> 
> Can we somehow detect from the software whether we have an unprotected AP set up?

Just with basic esp framework functions. 
```cpp
// required include
extern "C"
{
#include <esp_wifi.h>
}

// in code
wifi_config_t conf_current;
esp_wifi_get_config(WIFI_IF_AP, &conf_current);

if(conf_current.ap.authmode==WIFI_AUTH_OPEN){
// unprotected
}

Rotzbua, 2021-09-23:
The problems should be fixed with Arduino IDE ESP32 board framework `2.0.0`. Seems that older `1.0.x` wont get a fix https://github.com/espressif/arduino-esp32/issues/5351#issuecomment-925099042 .

Platformio is pending: https://github.com/platformio/platform-espressif32/issues/619