prampec/IotWebConf #189: exception (28) on example IotWebConf03TypedParameters

olonsoft, 2021-04-26:
Today I tried to test the example IotWebConf03TypedParameters
It compiled ok (platformio) but it could not run:
` ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

Starting up...
Config version: dem3
Config size: 428
Wrong config version. Applying defaults.

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40219c98 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000001 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffcf0 end: 3fffffc0 offset: 0190
3ffffe80:  3ffe8715 400042db feefeffe 40213a3c  
3ffffe90:  40004b31 3fff0fa4 000001b0 40201689
3ffffea0:  4020ab5c 3fff013c 3ffe8713 402016a1
3ffffeb0:  4020ab5c 3fff013c 3ffe8713 40214111
3ffffec0:  3ffefcc8 00000028 3ffefba4 40214111  
3ffffed0:  3ffefcc8 3ffefd14 3ffefd4c 40214111
3ffffee0:  3ffefcc8 3ffefd14 3fff013c 402076c3
3ffffef0:  3ffe8715 00000000 2f002e2e 4020d9f1
3fffff00:  3ffe86fe 4020df5d ffffffff 3fff01b8
3fffff10:  4020ab5c 00000001 00000002 40213a3c
3fffff20:  40213a50 3fff013c 3ffefcc8 40207750
3fffff30:  00000000 00000000 00000000 40208b7d  
3fffff40:  00000000 00000000 00000000 00000000
3fffff50:  40213a50 3fff013c 3ffefcc8 40203276
3fffff60:  402013e0 feefeffe 40213a78 40213a64
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3fff01b8
3fffffa0:  3fffdad0 00000000 3fff0178 4020c864  
3fffffb0:  feefeffe feefeffe 3ffe84e4 40100e1d
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------`

prampec, 2021-04-26:
Is this ESP32 or ESP8266?
You might also want to provide tool-chain version used at compile.

olonsoft, 2021-04-26:
```
Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.3) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting IotWebConf03TypedParameters.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <IotWebConf2>
|   |-- <DNSServer> 1.1.1     
|   |   |-- <ESP8266WiFi> 1.0 
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0 
|   |-- <ESP8266WiFi> 1.0     
|   |-- <EEPROM> 1.0
|   |-- <ESP8266mDNS> 1.2     
|   |   |-- <ESP8266WiFi> 1.0 
Building in release mode
.
.
.
Linking C:\programming\pio\.pioenvs\nodemcuv2\firmware.elf
Building C:\programming\pio\.pioenvs\nodemcuv2\firmware.bin
Retrieving maximum program size C:\programming\pio\.pioenvs\nodemcuv2\firmware.elf
Checking size C:\programming\pio\.pioenvs\nodemcuv2\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  42.7% (used 35004 bytes from 81920 bytes)
Flash: [===       ]  33.6% (used 351340 bytes from 1044464 bytes)
Creating BIN file "C:\programming\pio\.pioenvs\nodemcuv2\firmware.bin" using "C:\Users\irakl\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and "C:\programming\pio\.pioenvs\nodemcuv2\firmware.elf"  
=========================================== [SUCCESS] Took 58.03 seconds ===========================================
```