prampec/IotWebConf #190: Request Handler not found

Razsmithii, 2021-04-28:
I´m using this really cool tool for some monthes now and tried to integrate it into to another, more complex sketch. Connecting to WIFI works, but it´s not possible to open the Webserver on own AP nor connected to another wifi.
I use exactly the same fragemts of code in a single sketch (example3: CustomParameters) without having any trouble.
The error code i get via Serial Debug is:
dhcps: send_offer>>udp_sendto result 0
[E][WebServer.cpp:617] _handleRequest(): request handler not found

What am i doing wrong?
Is this the correct forum to answer this question?
If not, please be indulgend with me :-)

prampec, 2021-04-28:
Please re-check if server URL handlers are configured properly, as seen in the examples. E.g.:
  server.on("/", handleRoot);
  server.on("/config", []{ iotWebConf.handleConfig(); });
  server.onNotFound([](){ iotWebConf.handleNotFound(); });

It is very hard to say anything without seeing your code.

Razsmithii, 2021-04-29:
Thanks for you fast answer.
I double checked - URL Handlers are configured correctly.
It seems Bluetooth Serial combined with Webserver is the problem. 
As soon es "ESP_BT.begin()" is not called, everything works as it should be.
In some of my other sketches it works togther (BT and Wifi)
Are there already any known similar problems?
As i see, this coulb be a hardware problem caused of using the same antenna for BT and WIFI.

iw2lsi, 2021-06-25:
Hi Razsmithii,

   I'm facing the same problems while using ESP BT (classic)... did you found a solutions at the end ?

        tnx,

                Giampaolo