prampec/IotWebConf #191: Flash with default WiFi credentials

qroac, 2021-04-29:
I'd like to use IotWebConf on some ESPs in my dedicated IoT network.
But because my workstation does not have WiFi I'd have to connect every one of them from my smartphone, entering a long cryptic wifi password each single time.

That is why I'd like to flash the ESPs with a predefined WiFi SSID and PSK.  
Is that possible with IotWebConf?

prampec, 2021-04-29:
I was thinking about the same problem some days ago. I'm also thinking about a solution, where the configuration may arrive via JSON on either the file upload, MQTT message or by a file located in the flash area. But that is something migth to be done in the future.
What you can do, is that after init() fill out your parameters with defaults by your code, and do a configSave().

But this is not a documented and tested, further more not recommended way of handling e.g. passwords.

Also note, that there is a feature request about offline state (thus not using WiFi at all), that will be also useful for your application.

saffetblt, 2021-05-16:
Hi, can you show me how to do this? @prampec

prampec, 2021-05-16:
The following steps needs to be done:
1. Store a file on the flash area.
2. Read that file, and parse it.
3. Apply values to iotWebConf.

Let's do it from bottom to top:

3 - You can store update any value in iotWebConf, as mentioned in the HackingGuide: https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#accessing-system-properties

Note, that with typedParameters you can do a direct value-access for primitive types, something like ```param->value = myIntValue```.

2 - You can read files from the flash area (SPIFFS) with the provided methods:
- https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html

You might want to store your data in JSON format. To parse that, you might want to utilize the ArduinoJson library.

1 - To upload your config file to the flash area you might want to use PlatformIO provided method, to use a "data" folder for storing files to upload. Details on that: https://docs.platformio.org/en/latest/platforms/espressif8266.html#uploading-files-to-filesystem

As PlatformIO usually uses external script for assessing the MCU-s, it is very likely, that there is a python script for uploading files manually to the SPIFFS area.