prampec/IotWebConf #193: ESP32-CAM Update server fails

gb0101010101, 2021-05-09:
Using the IotWebConf04UpdateServer.ino example on ESP32-Cam (AI-Thinker ESP32-CAM board) and it fails with backtrace. I have tested the exact same code using ESP32 Dev Kit C board and it works. Having difficulty working out the problem. Can anyone provide info on where I should investigate to find the problem?

```
Backtrace: 0x4008efe0:0x3ffb1970 0x4008f211:0x3ffb1990 0x400eb19d:0x3ffb19b0 0x4008249f:0x3ffb19d0 0x400dcced:0x3ffb19f0 0x400dc4fc:0x3ffb1a10 0x400dc6fb:0x3ffb1a30 0x400d2559:0x3ffb1a50 0x400d398f:0x3ffb1a90 0x400db4f6:0x3ffb1ab0 0x400d73fe:0x3ffb1ae0 0x400d8091:0x3ffb1b10 0x400d8fbd:0x3ffb1cf0 0x400db6c6:0x3ffb1f00 0x40155555:0x3ffb1f50 0x400d4764:0x3ffb1f70 0x400d261e:0x3ffb1f90 0x400dde29:0x3ffb1fb0 0x4008af7d:0x3ffb1fd0
```

```
Decoding stack results
0x4008efe0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008f211: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400eb19d: is_safe_write_address at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 118
0x4008249f: spi_flash_erase_sector at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/spi_flash/flash_ops.c line 196
0x400dcced: EspClass::flashEraseSector(unsigned int) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/Esp.cpp line 305
0x400dc4fc: UpdateClass::_writeBuffer() at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/Update/src/Updater.cpp line 192
0x400dc6fb: UpdateClass::write(unsigned char*, unsigned int) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/Update/src/Updater.cpp line 297
0x400d2559: std::_Function_handler   >::_M_invoke(const std::_Any_data &) at /home/greg/Arduino/libraries/IotWebConf/src/IotWebConfESP32HTTPUpdateServer.h line 147
0x400d398f: std::function ::operator()() const at /home/greg/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/functional line 2271
0x400db4f6: FunctionRequestHandler::upload(WebServer&, String, HTTPUpload&) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WebServer/src/detail/RequestHandlersImpl.h line 89
0x400d73fe: WebServer::_uploadWriteByte(unsigned char) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WebServer/src/Parsing.cpp line 298
0x400d8091: WebServer::_parseForm(WiFiClient&, String, unsigned int) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WebServer/src/Parsing.cpp line 433
0x400d8fbd: WebServer::_parseRequest(WiFiClient&) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WebServer/src/Parsing.cpp line 199
0x400db6c6: WebServer::handleClient() at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/WebServer/src/WebServer.cpp line 302
0x40155555: iotwebconf::StandardWebServerWrapper::handleClient() at /home/greg/Arduino/libraries/IotWebConf/src/IotWebConf.h line 133
0x400d4764: iotwebconf::IotWebConf::doLoop() at /home/greg/Arduino/libraries/IotWebConf/src/IotWebConf.cpp line 575
0x400d261e: loop() at /tmp/arduino_modified_sketch_732263/IotWebConf04UpdateServer.ino line 94
0x400dde29: loopTask(void*) at /home/greg/.arduino15/packages/esp32/hardware/esp32/1.0.2/cores/esp32/main.cpp line 19
0x4008af7d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```
Attached ZIP archive or example code, bin, and elf files.
[IotWebConf04UpdateServer.zip](https://github.com/prampec/IotWebConf/files/6446833/IotWebConf04UpdateServer.zip)

In example code I just commented out `STATUS_PIN` and `CONFIG_PIN` code. Using the same exact code on ESP32 Dev Kit C board does work.

Thanks for any help in debugging this problem.

gb0101010101, 2021-05-09:
Sorry forgot to include the step by step info:

1. Compile & upload code using Arduino IDE or PlatformIO (via VScode)
2. Connect to 192.168.4.1 and set WiFi credentials and set `iwcApPassword` to "password"
3. IOT device connects to WiFi successfully (mode 4)
4. Open IOT web page (in my case http://192.168.1.127/)
5. Click on "Configure Page" link
6. Click on "Firmware Update" link (http://192.168.1.127/firmware)
7. Click on "Browse" button.
8. Select "IotWebConf04UpdateServer.ino.bin" file in the compiled directory (file attached previously).
9. Click "Update" button.
10. Get the failed update response with Backtrace.

gb0101010101, 2021-05-09:
Serial console output after configuring WiFi:
```
Starting up...
Config version: dem1
Config size: 165
Loading configurations
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'testThing'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'MY_SSID'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
\-- [hidden]
Ready.
State changing from: 0 to 2
Setting up AP: testThing
Use password: <hidden>
AP IP address: 192.168.4.1
AP timeout (ms): 30000
State changed from: 0 to 2
Connection to AP.
Disconnected from AP.
State changing from: 2 to 3
Connecting to [MY_SSID] (password is hidden)
WiFi timeout (ms): 30000
State changed from: 2 to 3
WiFi connected
IP address: 192.168.1.127
State changing from: 3 to 4
Accepting connection
State changed from: 3 to 4
Configuration page requested.
Rendering parameters:
[iwcSys]
|-- 'iwcThingName' with value: 'testThing'
|-- 'iwcApPassword' with value: <hidden>
|-- [iwcWifi0]
|   |-- 'iwcWifiSsid' with value: 'MY_SSID'
|   \-- 'iwcWifiPassword' with value: <hidden>
\-- 'iwcApTimeout' with value: '30'
[iwcCustom]
Form validation result is: positive
Updating configuration
[iwcSys]
|-- 'iwcThingName' with value: 'testThing'
|-- 'iwcApPassword' with value: <hidden>
|-- [iwcWifi0]
|   |-- 'iwcWifiSsid' with value: 'MY_SSID'
|   \-- 'iwcWifiPassword' with value: <hidden>
\-- 'iwcApTimeout' with value: '30'
[iwcCustom]

iwcThingName: testThing
iwcApPassword: <updated>
iwcWifiSsid: MY_SSID
iwcWifiPassword: <updated>
Config version: dem1
Config size: 165
Saving configuration
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'testThing'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: 'MY_SSID'
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
\-- [hidden]

Configuration page requested.
Rendering parameters:
[iwcSys]
|-- 'iwcThingName' with value: 'testThing'
|-- 'iwcApPassword' with value: <hidden>
|-- [iwcWifi0]
|   |-- 'iwcWifiSsid' with value: 'MY_SSID'
|   \-- 'iwcWifiPassword' with value: <hidden>
\-- 'iwcApTimeout' with value: '30'
[iwcCustom]
abort() was called at PC 0x400eb19d on core 1

Backtrace: 0x4008efe0:0x3ffb1970 0x4008f211:0x3ffb1990 0x400eb19d:0x3ffb19b0 0x4008249f:0x3ffb19d0 0x400dcced:0x3ffb19f0 0x400dc4fc:0x3ffb1a10 0x400dc6fb:0x3ffb1a30 0x400d2559:0x3ffb1a50 0x400d398f:0x3ffb1a90 0x400db4f6:0x3ffb1ab0 0x400d73fe:0x3ffb1ae0 0x400d8091:0x3ffb1b10 0x400d8fbd:0x3ffb1cf0 0x400db6c6:0x3ffb1f00 0x40155555:0x3ffb1f50 0x400d4764:0x3ffb1f70 0x400d261e:0x3ffb1f90 0x400dde29:0x3ffb1fb0 0x4008af7d:0x3ffb1fd0
```

gb0101010101, 2021-05-10:
After much research and testing I figured out the problem. It is not specific to this project so I am closing the issue.

The problem is caused by the default partition scheme used by ESP32-CAM which uses "Huge App" layout and does not leave any space for the OTA update to be written. For some reason this board does not come with predefined partitions schemes like most other boards.

The solution was to change to custom partition scheme by creating a `partitions.csv` file with the following contents.
```
# Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x5000,
otadata, data, ota, 0xe000, 0x2000,
app0, app, ota_0, 0x10000, 0x1E0000,
app1, app, ota_1, 0x1F0000, 0x1E0000,
spiffs, data, spiffs, 0x3D0000, 0x30000, 
```

If using Arduino IDE newer than 1.0.5 you just need to save the `partitions.csv` file to your sketch folder and it should be used by default.
If using PlatformIO then save the `partitions.csv` file to your `root` project folder (not `src` folder) and modify your platformio.ini adding:
```
board_build.partitions = partitions.csv
```
And Voila; OTA works!

closed 2021-05-10 by gb0101010101

prampec, 2021-05-10:
Wow! I needed some research to figure this out. Thank you for the feedback!

gb0101010101, 2021-05-18:
Glad I could help and thanks for the great work on this project. I use it on many Arduino boards (ESP8266 & ESP32) for my home automation with MQTT. Easily saving "Custom Parameters" to EEPROM, with UI to change them, has really made creating custom IOT solutions much easier for me.