prampec/IotWebConf #195: Lookup parameter by name, update value by hand

sacculus, 2021-05-17:
Hi
I have two questions.
First - is it possible to lookup some parameter by its internal name?
Second - is it possible to update values of parameters and save them in NVM at general runtime phase, not when portal running?

My device connected to cloud and I need to update some parameters with values from configuration channel of cloud at runtime.
I've look at examples but couldn't found such case.

Thank you!

prampec, 2021-05-17:
For quiestion#1:
Theoretically you can iterate through all the groups to find a parameter name, or ID. This feature is not exposed currently, but might be done, I'm not sure about the visibility of the fields and methods involved.

For question#2:
Please consult [Accessing system properties](https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#accessing-system-properties)
The same way you can modify any property.

sacculus, 2021-05-18:
Hi
Thank you for reply!
Q1: As temporary solution I use array which relates strings of property names and properties instances. But this not very comfortable - it is need to synchronise set of properties and array records each time when set of settings were updated.

Q2: Is this right way to assign property value like this (properties are addressed as void* so i use type cast)?
At this moment I've tested this method only with string values.

strncpy(((iotwebconf::TextTParameter<STRING_LEN>*)p_property_instance)->value(), "SOME STRING", SOME_LEN);

Is it right for numeric type value?
((iotwebconf::IntTParameter<int32_t>*)p_property_instance)->value() = SOME_INT32;

And one more question - is UIntTParameter class supports properties like min/max etc? When I tried build I've got an error.

PSS. Sorry for my dumb questions - I'm not very strong in C++...

prampec, 2021-05-18:
I'm also not very good at C++, but when you are using TypedPatameters (...TParameter) suddenly thing should go much easier.
In your case you should call p_property_instance->update(String("SOME STRING")), as you do not really know the type of the actual parameter being updated. (If you know the type however, your code is more or less correct. I would still use Update for updating text, as the length might not be known, or usage can be redundant.)

Min/max for UIntTParameter should be available, but this part is not well tested.

sacculus, 2021-06-03:
Hi
First question I realised with addition array for relation.
Second - works as expected. It is posible to update values by hand.

So I close issue.

Thanks!

closed 2021-06-03 by sacculus

sacculus, 2021-06-04:
Maybe my decision will help for somebody, so I share it.

At first small preamble.
I use esp32 for autonomous sensor, connected to Google Cloud IoT Core via MQTT. Main expectations is posibility to control system parameter remotely via cloud.
I use IotWebConf library for initial configuration of my device. But also I use it for update parameters at runtime with values arrived in json from config topic from MQTT.
For assotiation between parameters instances and its names (first question) I use array which relates them and also their types too.
I use typed parameters only ("...TParameter").

To reduce volume of example I show only key parts of code. If somebody will have interest of whole project I could share it on github. Write me.

PS. I had some trouble to use floating types, so I temporary avoided them by using of fractional values (millivolts, centidegree and so).

`
///////////////////////////////////////////
// Comment: Instances of parameters
///////////////////////////////////////////

// System
IotWebConfParameterGroup p_group_system = IotWebConfParameterGroup("system", "System configuration");

iotwebconf::IntTParameter<uint32_t> p_duty_cycle_interval_sec =
    iotwebconf::Builder<iotwebconf::IntTParameter<uint32_t>>("duty_cycle_interval_sec").
    label("Main loop duty cycle interval (sec)").
    defaultValue(10800).min(60).max(86400).
    step(60).placeholder("60..86400").build();

iotwebconf::TextTParameter<STRING_LEN_MAX> p_ntp_primary =
  iotwebconf::Builder<iotwebconf::TextTParameter<STRING_LEN_MAX>>("ntp_primary").
  label("Primary NTP server (IP or FQDN)").defaultValue("0.pool.ntp.org").build();

iotwebconf::TextTParameter<STRING_LEN_MAX> p_ntp_secondary =
  iotwebconf::Builder<iotwebconf::TextTParameter<STRING_LEN_MAX>>("ntp_secondary").
  label("Secondary NTP server (IP or FQDN)").defaultValue("1.pool.ntp.org").build();

// Google Cloud IoT
IotWebConfParameterGroup p_group_gloud = IotWebConfParameterGroup("gcloud", "Google Cloud IoT Core configuration");

iotwebconf::TextTParameter<STRING_LEN_MAX> p_gcloud_project_id =
  iotwebconf::Builder<iotwebconf::TextTParameter<STRING_LEN_MAX>>("gcloud_project_id").
  label("Project ID").defaultValue("").build();

iotwebconf::TextTParameter<STRING_LEN_MAX> p_gcloud_location =
  iotwebconf::Builder<iotwebconf::TextTParameter<STRING_LEN_MAX>>("gcloud_location").
  label("Location").defaultValue("us-central1").build();

iotwebconf::TextTParameter<STRING_LEN_MAX> p_gcloud_registry_id =
  iotwebconf::Builder<iotwebconf::TextTParameter<STRING_LEN_MAX>>("gcloud_registry_id").
  label("Registry ID").defaultValue("").build();

iotwebconf::TextTParameter<STRING_LEN_MAX> p_gcloud_device_id =
  iotwebconf::Builder<iotwebconf::TextTParameter<STRING_LEN_MAX>>("gcloud_device_id").
  label("Unique device ID (hostname)").defaultValue(thingName.c_str()).build();

iotwebconf::PasswordTParameter<STRING_LEN_MAX> p_gcloud_private_key =
  iotwebconf::Builder<iotwebconf::PasswordTParameter<STRING_LEN_MAX>>("gcloud_private_key").
  label("Device private key (in HEX form with :)").
  defaultValue("FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF:FF").
  build();

// Sensors
IotWebConfParameterGroup p_group_sensors = IotWebConfParameterGroup("sensors", "Sensors configuration");

// Sensor: Battery voltage (mV)
IotWebConfParameterGroup p_group_sensor_battery = IotWebConfParameterGroup("sensor_battery", "Battery voltage (ADC)");

iotwebconf::IntTParameter<uint8_t> p_battery_adc_samples =
    iotwebconf::Builder<iotwebconf::IntTParameter<uint8_t>>("battery_adc_samples").
    label("Number of samples to averaging values").
    defaultValue(16).min(1).max(255).
    step(1).placeholder("1..255").build();
[... 216 lines cut ...]

void systemConfigLoop() {
  if (systemConfigChangesPending) {
    Serial.println("System configuration changed, writing to NVM");
    iotWebConf.saveConfig();
    systemConfigChangesPending = false;
    // No more system task, can go asleep
    gcloudCanGoAsleep = true;
    // Manual change - noo need to reset device
    needReset = false;
  }
  if (systemConfigInitialized) {
    iotWebConf.doLoop();
    if (needReset)
    {
      Serial.println("Rebooting");
      ESP.restart();
    }
  }
  if (systemCanGoAsleep && gcloudCanGoAsleep && !systemConfigChangesPending) {
    goToSleep(p_duty_cycle_interval_sec.value());
  }
}

`