prampec/IotWebConf #196: WLAN credentials will not work

FluppFish, 2021-05-29:
Hello,
thank you for your cool Project!
But i have a problem with my wlan credentials.

SSID: Aquarium
PASS: X{dfe5%-+xI_2%s"fd0v2}=$9{]z#S?0X@8qERT54.s6G5W>\GhB*(&$?j6=2fw

There will no login at my AP (AVM Fritz!Box 7490)

If i use easyer login data like
SSID: Aquarium
PASS: asd123ASD
it will work.

Do you have a solution for that?

I tried to escape the " and the \ 
X{dfe5%-+xI_2%s\"fd0v2}=$9{]z#S?0X@8qERT54.s6G5W>\\GhB*(&$?j6=2fw

but this also will not work...

(Hint: This will not by my real password - only the special charters and the length are the same)

What i found in between:
Max size of the html password field is 32 charters...
https://s20.directupload.net/images/210528/thotysh7.jpg
Is there a way around?

I thested the file from MakeMagazinDE and it worked with hard coded credentials. I only had to escape the " and the \.
https://github.com/MakeMagazinDE/Node-RED-Special/blob/master/02-MQTT/mqtt_pub.ino
So the ESP8266WiFi.h is able to handle this kind of psk.

Thank you for your help...
Best regards
Markus

prampec, 2021-05-31:
Yes, this is a known issue with IotWebConf. As noted in #62 current password length is set to 32 character, that turns out to be out of the standard 64 characters.
The workaround mentioned there, is to set a custom password length for your project. As described here: https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#compile-time-configuration

Special characters should not cause any trouble of IotWebConf.

FluppFish, 2021-05-31:
Could you please explain me how to solve this?
Iam new in programming an esp8266 - and only wantet to flash a programm which are using your tool...

prampec, 2021-06-01:
Did you read the paragraph I have suggested?
If so, your configuration should be:
build_flags = -DIOTWEBCONF_PASSWORD_LEN=65

FluppFish, 2021-06-02:
Sorry in your link i could not find the "-DIOTWEBCONF_PASSWORD_LEN" option.
But thank you very much - this will work for me.
I am happy now ;-)

closed 2021-06-02 by FluppFish

prampec, 2021-06-02:
For the record:
All "compile time" configurations of IotWebConf are gathered in IotWebConfSettings.h .
So if you are looking for an option to tune, first check this file for the possible options.