prampec/IotWebConf #199: Connect to SSID without password

YiannisBourkelis, 2021-06-08:
Hello, how can I connect to an AP SSID that does not require a password? Currently the web interface does not allow a blank WiFi password.

prampec, 2021-06-08:
Right. IotWebConf is not intended to work on unsafe environment. IotWebConf exposes it's configuration on the network, so I cannot recommend any ways to do this.
If you want to work on an open network, you must know what you are doing. In this case you need to find your own way to do it.
Sorry!

YiannisBourkelis, 2021-06-08:
We are using your library in an environmental station and works pretty well. Now we are about to deploy some stations in the city and these stations are going to use the city public wifi to send the data. This WiFi is open (no wifi password) and for added security uses client isolation, so there is no way a user to intercept another user traffic. In that way there is no security issues for the IotWebConf. Even so, we are storing no sensitive data to it. 

To check if IoTWebConf can connect to open networks, I changed one line of code in file https://github.com/prampec/IotWebConf/blob/master/src/IotWebConf.cpp , in line 93 I added:
this->_wifiParameters._wifiPassword[0] = '\0';
just before:  
this->_apTimeoutMs = atoi(this->_apTimeoutStr) * 1000;
and it can connect to the open network without any issues.

I would like to suggest the addition of an option in the web interface of IotWebConf:
1. Allow empty wifi passwords. If the wifi password is empty, it will connect to the open network without using a password
or
2. A boolean option for using a wifi password or not.

Maybe I can give it a try to implement it if it is ok with you.

prampec, 2021-06-08:
I encourage you to fork this library and modify it to your needs, but I'm not willing to sacrifice security over additional options.
If you want to use empty passwords you want to check the validateForm() method.

YiannisBourkelis, 2021-06-09:
I understand your security concerns. We will take a look on the validateForm() method when we will have some time to see what we can do with it. Thank you.