prampec/IotWebConf #200: OTA update w/o password ?

iw2lsi, 2021-06-14:
Hi,

  thanks for the awesome library! I'm moving from WiFiManager and I must say that iotWebConf is simpler to use and much more reliable.

  I have a question... am I wrong or the webupdate is not asking (at least when in station mode) for credentials ? 

I am following the provided example... 

// Define how to handle updateServer calls.
   iotWebConf.setupUpdateServer(
      [](const char* updatePath) { httpUpdater.setup(&webserver, updatePath); },
      [](const char* userName, char* password) { httpUpdater.updateCredentials(userName, password); } 
      );

but I'm probably doing something wrong...

      Best Regards

               Giampaolo

prampec, 2021-06-14:
Update server must have the same protection as the config page. If this is not the case, it must be fixed.
Can you please confirm that this problem is not the same as this: #182 ?!

iw2lsi, 2021-06-14:
Hi...

   yes... it could be related to issue #182... I've also seens weird
behaviours with the AP appearing as "open" instead of "encripted"....

    I'll check it again...

              Giampaolo

iw2lsi, 2021-06-14:
Hi,

  I've done some test also with latest (on platformIO) upstream libs...
same behaviour...

1) the first time while trying to access the AP I was asked for the default
hard-coded psw
2) I've changed the "AP Password field", set the SSID and psw for the local
network and rebooted
3) during the 30s startup in AP mode, I was not able to connect to the AP
with the given new password... I've rebooted and suddently the AP appears
as OPEN.
4) accessing the device  using the station IP (192.168.1.174) seems not to
require any password... I don't now if this is expected or not.

Attached below is my log...

  Best regards

       Giampaolo

-----

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

MD240 WiFi/BT serial bridge v0.0.13c

Neopixel queue successfully created...
Neopixel task started...
[E][WiFiSTA.cpp:221] begin(): connect failed!

Starting TCP COM Server #1
Starting TCP COM Server #2

Starting web config service...

Config version: md01
Config size: 165
Wrong config version. Applying defaults.
[iwcAll]
|-- [iwcSys]
|   |-- 'iwcThingName' with value: 'MD240-AP_63accd98'
|   |-- 'iwcApPassword' with value: <hidden>
|   |-- [iwcWifi0]
|   |   |-- 'iwcWifiSsid' with value: ''
|   |   \-- 'iwcWifiPassword' with value: <hidden>
|   \-- 'iwcApTimeout' with value: '30'
|-- [iwcCustom]
\-- [hidden]
Starting Bluetooth (classic) SPP service as 'MD240-SPP' with PIN '1412'
configuration done...

AP password was not set in configuration
State changing from: 0 to 1
Setting up AP: MD240-AP_63accd98
With default password: <hidden>
AP IP address: 192.168.4.1
AP password was not set.
WiFi SSID was not set.
Will stay in AP mode.
State changed from: 0 to 1
Connection to AP.
[... 81 lines cut ...]
[E][WebServer.cpp:633] _handleRequest(): request handler not found
Request for www.msftconnecttest.com redirected to 192.168.4.1:80
[E][WebServer.cpp:633] _handleRequest(): request handler not found
Request for ipv6.msftconnecttest.com redirected to 192.168.4.1:80
Disconnected from AP.
State changing from: 1 to 3
Connecting to [MIFASTWEB2-4] (password is hidden)
WiFi timeout (ms): 30000
State changed from: 1 to 3
WiFi connected
IP address: 192.168.1.174
State changing from: 3 to 4
Accepting connection
callback wifiConnectionCallback invoked w/ state 4
State changed from: 3 to 4
Configuration page requested.
Rendering parameters:
[iwcSys]
|-- 'iwcThingName' with value: 'MD240-AP_63accd98'
|-- 'iwcApPassword' with value: <hidden>
|-- [iwcWifi0]
|   |-- 'iwcWifiSsid' with value: 'MIFASTWEB2-4'
|   \-- 'iwcWifiPassword' with value: <hidden>
\-- 'iwcApTimeout' with value: '30'
[iwcCustom]

prampec, 2021-06-16:
Thank you very much for the investigation. This might help others facing with the same issue. Let's leave this thread open in case new information drops in.