prampec/IotWebConf #201: TypedParameters and tables of data

societyofrobots, 2021-06-26:
What would be the best way to store a large table of data (20x5) using TypedParameters?

If there is no easy way, it's a recommendation to add for the next version. =)

prampec, 2021-06-27:
For structs, I would use parameter groups as seen in example IotWebConf14GroupChain.
To store large number of structs I would use the same technique, but store the groups in an array. An example for that can be found in IotWebConf15MultipleWifi:
```
iotwebconf::ChainedWifiParameterGroup chainedWifiParameterGroups[] = {
  iotwebconf::ChainedWifiParameterGroup("wifi1"),
  iotwebconf::ChainedWifiParameterGroup("wifi2"),
  iotwebconf::ChainedWifiParameterGroup("wifi3")
};
```

For tables I would form structs for rows e.g. in your example a row with 5 values. And then form an array of rows containing 20 rows.
(Of course as is, it will not be visualized as a table, but blocks of rows. Also note, that this answer has nothing to do with TypedParameters; there might be a way to visualize (and store) blocks of data with typed parameters, but currently there is no recommendation for that.)

societyofrobots, 2021-06-27:
Excellent!

What if I wanted to do a log that gets appended to occasionally, how would I best do that?

prampec, 2021-06-28:
I'm not sure what kind of log you are talking about.
IotWebConf logs to Serial by default. You can disable logging, but this is a compile time decision, so it is not something you can change runtime.

societyofrobots, 2021-06-28:
I meant a log saved on the ESP32 itself, not serial.

For example, I am planning to create an access log file that'll store a date stamp of when people log in and the actions taken. Any user can then view the logs after logging in.

Although on second thought, I'm thinking maybe a SPIFFS or FFat library would be more appropriate.

prampec, 2021-06-28:
I recommend logging into an InfluxDB (via MQTT) installed on your central server.
I don't recommend using flash for extensive writes.

societyofrobots, 2021-07-25:
MQTT won't work as I require secure storage of the log, and encryption isn't easy.

Writes will not be extensive. At ~600 bytes written/day, 274kb free dynamic memory, and 10,000 minimum writes to failure, that's >12,000 years to failure.

Ideally it'd save to a .txt file, which then can be accessed via AP mode. I'll continue investigating and report back.

societyofrobots, 2021-08-29:
What I ended up doing is storing the table in one big SPIFFS file.

If I need to manipulate the data:
1) I read the file into a String
2) Change what's needed in the String
3) Store it in a temp file
4) Delete the original temp file
5) Rename the temp file to the original file name
6) Pray that power doesn't cut out during the delete phase, cause I'm too lazy to do it properly with flags....

It's only a dozen re-writes or so per year, so not worried.

closed 2021-08-29 by societyofrobots