prampec/IotWebConf #203: TaskScheduler library compatibility

coelner, 2021-06-28:
I try to combine this library with https://github.com/arkhipenko/TaskScheduler/

You need to define a task which is done by this line: 
`Task iotwebconfLoop(1000, TASK_FOREVER, &(iotWebConf.doLoop));`
bit it fails with this: 
`ISO C++ forbids taking the address of a bound member function to form a pointer to member function.  Say '&iotwebconf::IotWebConf::doLoop' [-fpermissive]`
doing this:
`no matching function for call to 'Task::Task(int, int, void (iotwebconf::IotWebConf::*)())'`

http://www.parashift.com/c++-faq-lite/typedef-for-ptr-to-memfn.html

And while struggling about this, how often the doLoop() call needs to be called?

prampec, 2021-06-28:
Basically this is a design issue with the library you are using. (Not that mine is better in this respect https://github.com/prampec/arduino-softtimer)

The solution is to define a method on your own to wrap IotWebConf call, and pass your method reference.
Something like this.
```
void myMethod();
Task iotwebconfLoop(1000, TASK_FOREVER, &myMethod);

void myMethod()
{
  iotWebConf.doLoop();
}

coelner, 2021-06-28:
that was too obvious, because I did this with all other tasks :roll_eyes:

Thanks a lot!

closed 2021-06-28 by coelner