prampec/IotWebConf #204: Hook (callback) for AP mode timeout

sacculus, 2021-07-11:
Hi
Thanks a lot for this excelent library!
I use it with battery powered device. All works as expected but I faced one problem.
When WiFi not accessible library run sequentially station and AP modes in endless loop and this ate battery wery fast.
Pls, tell me how I can catch event when AP mode timed out? I'd call my routine to sent device asleep after first try of connection?

prampec, 2021-07-19:
Please check "Control on WiFi connection status change" section of the "Hacking Guide". https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md#control-on-wifi-connection-status-change
Also note, that there is a new way, where you can disable the WiFi radio completely manually. For that, you must check the ```IotWebConf16OffLineMode``` example.

closed 2021-07-19 by prampec

sacculus, 2021-07-20:
Hi
Excellent! Thank you!