prampec/IotWebConf #217: load JsonConfig does not work with checkbox

MatMat81, 2021-09-06:
tested with parameters:  '"true"', "true", "1", "'1'", "selected", "checked"
 ....
none switches the checkbox to true.
All other parameters can be set ...
....
         "iwcCustom": {
             "c_factor": {
                 "flParam": "55.25",
                 "checkParam2": true,
                 "checkParam": "true",
                 "chooseParam": "red"
             },
             "intFac": {
                 "inParam": "20"
             }

MatMat81, 2021-10-03:
i have found the solution:
"checkParam": "selected" 
is the key word for setting the checkboxes...

completed...

closed 2021-10-03 by MatMat81

prampec, 2021-10-03:
Yes. JsonConfig will be much nicer if TypedParameter approach can overtake regular parameters. In that case parsing should be done type based, meaning float parameters will accept numbers instead of string on Json.

MatMat81, 2021-10-04:
can you add the solution to the config.json example No.17 so that other people have no problem?

prampec, 2021-10-04:
I have added this note to the README of the example, because it should also be documented how a not selected value look like.