prampec/IotWebConf #220: a little help needed

Ricky6633, 2021-09-16:
Hello i have a few points 

1. How we do groups in groups? i readed in the documentation that its possible but couldnt find a example which utilizes it (a documentation of all api functions would be nice)

2. is it possible to create mouseover tooltips for parameters? (if not its a feature request)

3. is it possible to create parameters which are just for showing values? (so "readonly") or would it be better to create a custom 192.168.4.1/info page?

4. i also saw the example in the documentation to change the label of the system parameters but it didnt work, could i get a full working example?, im pretty sure im just missing some namespace stuff but i tried some and it didnt work

prampec, 2021-09-16:
1. In example 03 you see `group1.addItem(&intParam)`, now here you can also add a group reference.
2. You are free to modify HTML presentation of the config page as you can see in example 10. Parameter appearance however is a different thing. Here you need to perform some magic like the one you can find in IotWebConf12CustomParameterType. You might also need to store the tooltip text somewhere. Mouseover tooltips is not likely to be provided by IotWebConf, as the GUI is mostly optimized for smartphone use, where mouseover is not a thing.
3. Yes, I recommend to provide an info page. I usually use the landing page for that.
4. I believe this will do the trick: `iotWebConf.getApPasswordParameter()->label = myStaticString;` (Where myConstString should be a `const char[]` variable, or regular "my label" like string can also be used, if I'm right.

Ricky6633, 2021-09-16:
1. Like this ?  "group1.additem(&group2);"  i tried that before and it gives me an error "using IotWebConfParameterGroup = class iotwebconf::ParameterGroup {aka class iotwebconf::ParameterGroup}' has no member named 'additem'"
2. ah that makes sense, i didnt think about smartphones, thanks! 
3. ok i will use seperate page
4. iotWebConf.getApPasswordParameter()->label = "labeltest";  works without seperate char/string too, thanks :) is there somewhere a list of all parameter names? i search specially this one iotWebConf.systemParameters.label = "My Custom Label"; which seems outdated or something

Ricky6633, 2021-09-16:
4. nvm it is iotWebConf.getSystemParameterGroup()->label = "WLAN";

Ricky6633, 2021-09-16:
Are there some issues with smartphones? i just notices i cant connect with mine, 192.168.4.1 always times out, while its working fine on desktop at the same time 
Connecting to wlan works but 192.168.4.1 timesout

Ricky6633, 2021-09-17:
1. not sure what i did different but i tried group1.addGroup(&group2); after that group1.additem(&group2); again and it worked now... maybe i had some nasty typo while i tried the first time

i will close the issue since all questions are answered (for now) :) Thanks!

closed 2021-09-17 by Ricky6633