prampec/IotWebConf #221: How to change custom parameters in Code?

Ricky6633, 2021-09-21:
Hello i found another Issue which answered this particular question about TParameters but im unsure what syntax i need to use for "normal" parameters
(Btw are TParameters "recommended" ? or should i stick with the "normal" char ones?)
i tried different stuff including  iotwebconf::NumberParameter(customparameter)->valueBuffer());
i think im kinda close but i have no idea where to get the needed syntax-information from, unfortunaly there is barely documentation

Thanks for the awesome library btw its working quite flawlessly so far

prampec, 2021-09-21:
Can you please reference the issue you have mentioned?!
I personally only use TParameters exclusively. There are some problems with it: The code is a mess and needs to be cleaned up, the system parameters are not TParameters.
We have some special examples about custom parameters, have you checked those?
There is also a [Hacking guide](https://github.com/prampec/IotWebConf/blob/master/doc/HackingGuide.md), which is The Documentation for customization.

Ricky6633, 2021-09-21:
Yes i checked the Hacking Guide 
This is what the guide suggests 

    // -- Update Thing name
    strncpy(
      iotWebConf.getThingNameParameter()->valueBuffer,
      "My changed name",
      iotWebConf.getThingNameParameter()->getLength());
    iotWebConf.saveConfig();

to update values by hand, thats also working fine but for me just for systemparameters (because the guide has a list of the available (system) parameters), because i dont know the syntax for custom made (nonT) parameters

what needs "getThingNameParameter()" to be for custom parameters because when i just use the name of the parameter its not working

Also, is there a difference between ->valueBuffer and ->value ?
(should it not be always the valuebuffer to write data to and saving the config would write it to value?), Or is ->value just for TParameters like used in this issue -> https://github.com/prampec/IotWebConf/issues/195 ?

About TParameters, then it seems i made the right choice with sticking with nonT parameters for now

prampec, 2021-09-22:
Sorry, I still not understand the problem. What do you mean about using the name of the parameter? The example you have referenced here is about nonT parameters. And overall the documentation does not really mention TParameters, this is intended to be so not to convince developers.

->valueBuffer is the classic way, ->value() is the Typed-way to access the value of the parameter.

Ricky6633, 2021-09-22:
->valueBuffer is the classic way, ->value() is the Typed-way to access the value of the parameter.
Ah.. that makes more sense

Well i dont know how to explain it differently...
If i wanna change the thing name, i use this -> iotWebConf.getThingNameParameter()->valueBuffer

if i wanna change other systemparameters i use the corresponding "get" function

but if i have a custom parameter named "custom_parameter" what would be the "command" to get the valueBuffer of "custom_parameter"

like i said before i tried this already: iotwebconf::NumberParameter(custom_parameter)->valueBuffer());
but its not working because i used the wrong "syntax/command" -> NumberParameter(custom_parameter)

what do i need to type -instead- to get the valueBuffer of "custom_parameter" because there is no function like  "getThingNameParameter()" for custom made parameters

prampec, 2021-09-23:
Ohh, okay. :) Let's see.
All the system parameters has getter, as you are not able to see them anyway, they are private part of the IotWebConf.
Now, with all your own custom parameter, you can access them from your own code. E.g. in example 03 you can directly access floatParam, thus floatParam.valueBuffer() .
I believe, what you are looking for is something to find the parameter by ID. E.g. Parameter* myParameter = iotWebConf.findParameterById("myId1"); ... Unfortunately iotWebConf does not provide such a feature.
What you can do, is to build your own map with your parameter ID-s, and find your parameters in that map.

(But this request rings the bell for me to... I might want to reorganize the group-structure, so that Parameters does not have their own ID, but instead a key is required, when this parameter was added.)

Ricky6633, 2021-09-23:
floatParam.valueBuffer()

I just need this one above, basicly i just need to write uint8_t values to the two hidden parameters i created and if it works that way it would be fine, no need for special "id-search" or something like that

unfortunaly if i try "floatParam.valueBuffer()" it i get "expression cannot be used as a function" as compile error

Sorry btw, im still quite a noob so maybe i just dont get the obvious here :/

prampec, 2021-09-23:
Then what you are looking for is itoa() (https://www.cplusplus.com/reference/cstdlib/itoa/)
Where str is myParam.valueBuffer (or from example 03 intParamValue).
Please note the length-constraint mentioned in the manual above!

Ricky6633, 2021-09-24:
Ah thanks, just "myParam.valueBuffer" works, since its a uint_8 im unsure if itoa works since it takes a int instead, (and i forgot to mention that its actually a uint8_t [139] array from the bsec bosch library)

do i have to explicitly use the conversion to string to get iotWebConf going? (since i readed large string should be avoided)
or could i also convert the uint8_t array to a char array and simply write it with for example
"myParam.valueBuffer = (char*)uint_array" and save settings afterwords?
(or maybe even better memcpy, but i still have problems with the conversion of uint8_t to char :/ )

Ricky6633, 2021-09-24:
hmm since the uint8_t array i have to save has also a value from 0 to 255 for a byte it would be quite complicated to save the value (maybe there is a easy way i dont know tho..) but im wondering if it wouldnt be better to save them directly as uint8_t, would it be possible to save the two uint8_t i need to save on the "upper end" of the EEPROM, so i dont have to mess with librarys files? (i know the config_start parameter exists) but if i save them on the "upper end" (eg byte 3500 to 4096) of the eeprom would it work without conflicts?

Ricky6633, 2021-10-01:
I "fixed" the problem with converting the needed uint8_t to the char array everything seems to work now 

But i found out that i can also just use the created char`s instead of the corresponding .valueBuffer

so im wondering, what is the real benefit of using the valueBuffer? 

for example
Mode = Mode_P1_char; 
is working just fine, Mode_P1_char is the char which gets used by iotwebconf

or does that just work with READING the value but not writing to it?

MatMat81, 2021-10-04:
I have the same problem to read and write customer values by code.

3 functions would be needed to solve the problem for DIY-people 
for example:
> "float fcustom = LoadValue(CustomFloatParam, [DataType=float])"  //get from eeprom
> "SaveValue(CustomFloatParam, 12.34)"  //save to eeprom
> "UpdateValue(CustomFloatParam)  //Update Value in WebPage

... unfortunately there is no such thing so non-professionals get no further here

prampec, 2021-10-04:
When you init() iotWebConf it will load all the configuration tree specified from the EEPROM (in case it is already there). You can get a value form your value buffer, as you have specified it yourself. (As shown in the method handleRoot() of example IotWebConf03CustomParameters.) The same way you can write values to your valueBuffer directly. (Note, that ```iotWebConf.saveConfig();``` must be done to apply changes in EEPROM.)

valueBuffer is always a string (char[]). There are several ways to convert a specific type to string and back from string to a specific type in C language. I bet Google helps on that.

If you are keen to use native types however, you should use the experimental TypedParameter approach. (It is experimental because it might lacks some features.) There you can directly read a value in a native type (as shown in IotWebConf03TypedParameters), and write value with e.g. ```piParam.value() = 3.14;``` (Again configSave() call is required after a change.)