prampec/IotWebConf #222: Easy way to check for active AP mode?

Ricky6633, 2021-10-01:
Hello, is there a way to check easly for active AP Mode? 
I wanna show two different symbols on a display, one for active AP mode and one for Connected Wifi
ConnectedWifi is no problem since i can use the WifiConnectedCallback to set a bool but im unsure how to switch back the bool variable if ap mode is active again ( if wifi connection failed for some reason )

prampec, 2021-10-01:
You can define Handlers (callbacks) for both AP and Wifi. The only thing you need to take care is to call the actual handlers before or after your custom tasks are done.
Another way to do this, is to frequently monitor the getState() and change you display accordingly.

Ricky6633, 2021-10-01:
Thanks alot for the fast reply

`void CheckWifiStatus() {
  if (iotWebConf.getState() == 2) {
    wlanConnected = false;
  } else if (iotWebConf.getState() == 4) {
    wlanConnected = true;
  }
}`

did the trick
wlanConnected = false means AP is on, true means Wifi is connected :)

prampec, 2021-10-01:
Instead of 2 / 4 you should use iotwebconf.ApMode / iotwebconf.OnLine enum constants.

prampec, 2021-10-01:
> wlanConnected = false means AP is on, true means Wifi is connected

And what about the other four states? ;)

Ricky6633, 2021-10-01:
Would you have a example of how to use the enum constant instead of the "raw values" ?

I actually just need to know if ap mode is on OR wlan is connected (for switching the coresponding symbol on display) or is there a benefit i dont see?

hmm maybe also checking if none is "active" to hide the symbol, but in normal operation this should never happen right?

Ricky6633, 2021-10-01:
Boot, NotConfigured, ApMode, Connecting, OnLine, OffLine

Ah well, i found the states, im wondering what time "NotConfigured" and "OffLine" is active

i updated my code as follows so far

  if (iotWebConf.getState() == iotwebconf::ApMode) {
    lcd.write(3); //wlan_inverted
  } else if (iotWebConf.getState() == iotwebconf::OnLine == true && mqttClient.connected()) {
    lcd.write(4); //wlan_mqtt
  } else if (iotWebConf.getState() == iotwebconf::OnLine) {
    lcd.write(2); //wlan
  }