prampec/IotWebConf #223: IotWebConf on demand as configuration tool

ClemensGruber, 2021-10-01:
My ue case is a bit different than the scenarios descibed at https://github.com/prampec/IotWebConf#use-cases so perhaps someone can say if it is possible with IotWebConf or give some hints how to reach it. I had a look at different examples and the documentation but I have not a deep understanding till now. 

We have an open souce bee hive project with different hard- and software setups at [hiveeyes ](https://community.hiveeyes.org/) variants. For an implementaton with TTGO's T-Call I'd like to use IotWebConf for configuraion. Till now it is done in the Arduino sketch but this is not so usable, especally for people / in this case non-technical beekeeper. :-) 

So the story is: 

- We have a device that is solar powered with a buffer LiPo, power saving and deep sleep is a must have. 
- Normally the device 
-- wakes up from deep sleep, 
-- does some measurement, in this case weight of the bee hive, some temperatur and / or humidity sensors, later perhaps also audio for diagnostics,
-- and send this data via Wifi or GSM -- the T-Call is an SIM800 equipped ESP32 dev board -- to our InfuxDB / Grafana server
-- go in deep sleep again

For configuration -- and only for this task -- we need the IotWebConf mechanism with active Wifi and UI.

This should be **on demand**. That means 
- normally Wifi sould not be activated to save the less power of the sun harvested device 
- but even if the UI and Wifi runs not we need accesss to the variables modified and collected via IotWebConf 
- in case configuration mode / on demand is activated we need a **timeout**, in case someone forget to disconnect from the IotWebConf UI the device should stopp the UI and WiFi after a time and go in deep sleep, otherwise the hive scale will run out of battery. 

At the moment I have planed to call IotWebConf UI triggered by reset causes, so in case someone pressed reset the webconfig sould be called. 

My idea right now is to move `iotWebConf.doLoop();` out of `loop()` to a function that is only called after the reset cause "poweron reset" / or a button press or an other trigger for the web UI. 

In this function configuration is done and after submitting the config parameter it returns from this function to loop with the other staff above and Wifi should be deactivated to save power. 

Is this doable with iotWebConf or should I spent time for other things? ;-)

ClemensGruber, 2021-10-02:
In the GitHub issues and library examples I found `forceApMode(),` that will be helpful for the described scenaro. So I can omit joinig a Wifi network that's not there because we are using GSM/GPRS/LTE for the internet connection. 

Switching off Wifi can be done with `iotWebConf.goOffLine();` after configuration, see [IotWebConf16OffLineMode.ino](https://github.com/prampec/IotWebConf/blob/master/examples/IotWebConf16OffLineMode/IotWebConf16OffLineMode.ino). 

So remainig is the on demand quesiton. Is there a command to say with this condition only activate Wifi and run the AP? Or do I have ever to activate the AP mode and go immediately in offline mode? This would not be a good solution for battery powered devices.

amotl, 2021-10-03:
Dear Clemens,

it looks like @kugelkopf123 has a similar use case, see also #112:

> - To be able to completely do without the AP / STA Autoconnect thing in order to be able to use only the Configportal (not captive portal).
> - The possibility to prevent the automatic connection to the entered access point (STA).
> - Background: In order to save as much energy as possible.

With kind regards,
Andreas.

amotl, 2021-10-03:
> Or do I have ever to activate the AP mode and go immediately in offline mode?

Maybe the right method to invoke is `startupOffLine()` instead?

csiki2, 2021-11-07:
Hi,

I do something similar, the solution in short:
<pre>
  _webConf.skipApStartup();
  addParameters();

  const char* mode;
  if (tryLoadConfig) {
    if (!_webConf.loadConfig() && ...) return false;
...
    mode = "load config";
  } else {
...
    _webConf.init();
    _webConf.forceApMode(true);
    mode = "access point";
  }
</pre>