prampec/IotWebConf #224: upgrading to v2.0 core?

societyofrobots, 2021-10-01:
I spoke with you about this before, and you understandably didn't want to do the significant work needed to get core v2.0 compatibility.

Normally I'd be ok with just 1.0.6, but that version has an encryption bug that won't let me use anything but [TKIP which iOS complains about](https://github.com/espressif/arduino-esp32/issues/4805). v2 has fixed it.

Now that 2.0 is officially out for about a month, I was curious if you were to make the upgrade?

ClemensGruber, 2021-10-01:
I think it should runs with 2.0.0!

prampec, 2021-10-03:
Espressif 32 platform package v3.3.2 does not updates framework-arduinoespressif32 3.10006.210326 (1.0.6), so I cannot confirm that on PlatformIO.

atclarkson, 2021-10-05:
I'm having the same issue, is there a way to force 2.0.0 using platformio?