prampec/IotWebConf #234: wpa2 enterprise?

l33333on, 2022-01-07:
Hi, is there a possibility to connect my esp32 to a WPA2 enterprise wifi with radius authentication?

prampec, 2022-02-22:
I believe this is something depends on the ability of the ESP32. IotWebConf just gathers abilities of the ESPs and put a nice frame around it.

marcelstoer, 2023-11-09:
ESP32 Arduino does support connecting to WPA2 Enterprise APs. See https://github.com/JeroenBeemster/ESP32-WPA2-enterprise/blob/master/ESP32_WPA2enterprise.ino for an example. So, from what I understand we need either of two things:

- a way to skip IotWebConf WiFi initialization and do it yourself
- IotWebConf to do this transparently behind the scenes for us