prampec/IotWebConf #236: Password fields should have asterisk or dots instead of being empty

schmurtzm, 2022-01-09:
Hi,

I have noticed in web interface that when I'm changing a MQTT setting, the "AP password" and "WiFi password" are displayed as empty fields instead being filled of asterisks. 
This is not a typical behavior because it doesn't indicate that these passwords have already been set. Also it doesn't allow to empty these fields (unlike the others).

Thanks for this awesome library 😉

prampec, 2022-02-22:
Empty password is not allowed by design. IotWebConf tries to be safe.
HTML rendering of the password fields are done by the browser. IotWebConf does not try to modify the default behavior of the browsers in this manner. In my experience Android browsers displays dots for every character entered.
(Also note, that there is an option, when you double-tap on a password field, that it displays it's content.)

closed 2022-02-22 by prampec

schmurtzm, 2022-02-22:
Oh sorry I was not clear enough in my explanation, I think it should stay open...

I mean when you have already set your configuration, you go back to the UI and then it is like there is no password set : all the passwords fields are empty instead of containing a random number of points or asterixes.

So this behavior has 2 drawbacks : 
- it encourages the user to re-enter his password when he wants to change a parameter when it is not necessary.
- it is impossible to tell if the MQTT password has already been entered or not

ClemensGruber, 2022-04-11:
For me it's also a lack of usability. In case you have set up your device and go to the config page again, the WiFi password and AP password fields are emtpy. What does this mean? Is there no password set? Do I have to set the password again? A new password, or the allready stored password? Was tha password stored last time or was there an error. In case I save the configuraton untouching the mentioned password inputs, my thought as user are: What is saved an empty string as password? The las password, so it is untouched? Should be more clear by displaying dots in case a password is allready there or an empty field with an additional hint "enter password" or similar to make this more clear.

prampec, 2022-04-11:
I understand your concerns. HTML password input provides a _placeholder_ text, that will appear, when no values are provided. I can change this placeholder to be a different text to help users distinguish between entering new password or modifying exiting password.

What text do you recommend for the two cases? E.g. "Enter password" vs. "Password is hidden"?

prampec, 2022-04-11:
What about this? "Not set" vs "Skip if not changed"

prampec, 2022-04-11:
Btw. you are already free to modify myParameter.placeholder value to a text you would like to see if field is empty. The only thing is, that currently no separate placeholder for "empty, but already set" values.

schmurtzm, 2022-04-14:
HI thanks for your concern :)
I think that the best way still to put asterisks or dots when it is set because this is the most common way. 
The place holder can be an alternative, in this case your second proposition seems good to me.