prampec/IotWebConf #237: AP mode always available beside STA

alirezaimi, 2022-01-22:
Hi 
How can i use both station mode and AP mode together in this library with out loosing AP mode after some times on start up only ? STA+AP 
In other words how can i prevent disabling AP after some times that declared in this library ?

prampec, 2022-02-22:
Can you please form this question again?!
IotWebConf does not use AP and STA together, and currently there is no recommendation how to have this done.
AP and STA modes are alternating if cannot connect to a WiFi network in STA mode.

alirezaimi, 2022-04-26:
I want both of them in case that one of them goes down, i can connect to board . It's important that make softap available beside sta mode .

alirezaimi, 2022-05-22:
Hi
Is there any option to use STA and AP together !?  
I need to esp board connect to modem access point to use internet for ntp and then i connected to it whenever modem not available or when i do not know board ip in network ...

prampec, 2022-05-22:
The main idea of this library (besides of the conf. portal) is that iotWebConf will manage network connection (start in AP mode, tries to connect, fall back to AP when connection is lost, etc.)
What you want here is a special use-case, that unfortunately does not fit to this concept.

You can try and hack the network connection using example IotWebConf09CustomConnection as a template, but iotWebConf also does some housekeeping jobs, so I'm not sure how you can make it reliable. And certainly I myself cannot provide a stable solution for you for this use-case.

alirezaimi, 2022-05-23:
Thanks for your response , but when esp8266 have this feature why we don't use it !? why ?! softap for having both AP and SAT mode is one of base feature of this board and you just not seeing it !

prampec, 2022-05-23:
I see a bigger picture. Imagine a smart home with 20+ smart devices. 20 WiFi APs at a time would flood WiFi discovery. (Might also collide with each other.)
But you are right, this would be a nice option on iotWebConf.

prampec, 2022-05-23:
> whenever modem not available or when i do not know board ip

Whenever modem is not available iotWebConf will fall back to AP mode automatically, so you can access it. (The downside is that it tries to reconnect regularly and closes AP for that time.)

mDNS might be a solution in case you don't know the IP. (mDNS received some major fixes bugfix the latest release.)

But you are right: I also recommend to perform any configuration changes in AP mode, as it is known to be more secure than local WiFi network having untrusted devices connected to it.

Also: You might want to utilize a hardware button, that forces AP mode any time you are willing to connect to the device via AP mode. See method ```forceApMode()```

alirezaimi, 2022-05-23:
Yes, sure , but on that point user can choose flood or the other ways .
This is as an option would be nice choice for user , 
thank you very much .

alirezaimi, 2022-05-24:
I use this example code and 3 things(mDNS, Webupdate, softAP) that not work with `iotwebconf` library, works perfectly in this code :

    #include <ESP8266WiFi.h>
    #include <WiFiClient.h>
    #include <ESP8266WebServer.h>
    #include <ESP8266mDNS.h>
    #include <ESP8266HTTPUpdateServer.h>

    #ifndef STASSID
    #define STASSID "your-ssid"
    #define STAPSK  "your-password"
    #endif

    const char* host = "esp8266-webupdate";
    const char* ssid = "XXX";
    const char* password = "YYY";

    ESP8266WebServer httpServer(80);
    ESP8266HTTPUpdateServer httpUpdater;

    void setup(void) {

      Serial.begin(115200);
      Serial.println();
      Serial.println("Booting Sketch...");
      WiFi.mode(WIFI_AP_STA);
      WiFi.begin(ssid, password);

      while (WiFi.waitForConnectResult() != WL_CONNECTED) {
        WiFi.begin(ssid, password);
        Serial.println("WiFi failed, retrying.");
      }

      MDNS.begin(host);

      httpUpdater.setup(&httpServer);
      httpServer.begin();

      MDNS.addService("http", "tcp", 80);
      Serial.printf("HTTPUpdateServer ready! Open http://%s.local/update in your browser\n", host);
    }

    void loop(void) {
      httpServer.handleClient();
      MDNS.update();
    }

prampec, 2022-05-25:
Can you please be more specific! Hundreds of people using `iotWebConf` library and no one ever claimed about AP mode and WebUpdate failures. Sure mDNS has it's issues, but with the latest update I believe most of these are solved.